tsduck/tsduck #240: Astra NIT not decoded correctly by tstabcomp (or not stored in tstables output)

bobben1no, 2019-03-08:
I have a problem decoding ASTRA NIT table from 19.2E 12.606 H 22MS/s 5/6.  For some reason tstabcomp will only decompile NIT other section.  It seems like NIT actual section with table_ext 0x0001 is always ignored. NIT other section with table_ext 0x0003 is always presented in the XML output.  I'm not sure if it is happening inside tstabcomp of when the tables is stored in BIN file by tstables.

I have been looking at the TS packets in hex viewer and I notice that the NIT actual sections is having pointer_field which is much larger than the pointer_fields in the NIT other sections. 

The command I am running in windows batch is :
```
tsp -O file ASTRA_SI.ts -I dvb --satellite-number 2 --frequency 12603000000 --symbol-rate 22000000 --polarity horizontal --delivery-system DVB-S --fec-inner auto -P sifilter --pat --cat --pmt --nit --sdt --tdt -P skip 100 -P until --seconds 30
tstables --binary-output ASTRA_SI.bin ASTRA_SI.ts
tstabcomp --decompile ASTRA_SI.bin
```

Example NIT actual section start :
```
47 40 10 1F 94 FF FF FF FF FF FF FF FF FF FF FF   G@   ÿÿÿÿÿÿÿÿÿÿÿ
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF   ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF   ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF   ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF   ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF   ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF   ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF   ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF   ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ
FF FF FF FF FF FF FF FF FF 40 F0 20 00 01 D5 02   ÿÿÿÿÿÿÿÿÿ@ð   Õ 
02 F0 00 F0 13 04 06 00 01 F0 0D 43 0B 01 16 56    ð ð     ð C   V
00 01 92 A6 02 20 00 02 A5 8C DC 6E                         Ün  
```

Example  NIT other section start (most have pointer_field=4):
```
47 40 10 14 04 FF FF FF FF 41 F1 68 00 03 D9 00   G@   ÿÿÿÿAñh  Ù 
00 F0 2B 40 08 41 53 54 52 41 20 33 00 4A 07 0C    ð+@ ASTRA 3 J  
91 00 03 18 2E 07 5F 04 00 00 00 01 4A 07 0C 98       . _     J   
00 03 14 55 93 4A 07 0C 98 00 03 00 00 04 F1 30      U J        ñ0
0C 82 00 03 F0 0D 43 0B 01 17 39 00 02 35 B6 02       ð C   9  5  
99 00 02 0C 88 00 03 F0 0D 43 0B 01 18 56 00 02          ð C   V  
35 B6 02 99 00 02 0C 89 00 03 F0 0D 43 0B 01 18   5         ð C   
75 50 02 35 8E 02 99 00 03 0C 8E 00 03 F0 0D 43   uP 5         ð C
0B 01 19 73 00 02 35 B6 02 99 00 02 0C 91 00 03      s  5         
F0 0D 43 0B 01 20 31 50 02 35 85 02 75 00 09 0C   ð C   1P 5  u   
93 00 03 F0 0D 43 0B 01 20 70 50 02 35 81 02 75      ð C   pP 5  u
00 03 0C 95 00 03 F0 0D 43 0B 01 21                     ð C  !
```

I have a ts recording of 3Mbyte that can be shared but I'm not sure how to share it here.

lelegard, 2019-03-11:
Hi @bobben1no,

I have a dish pointing to Astra 19.2E and I got a similar capture file (using 12.604H as reported by [Lyngsat](https://www.lyngsat.com/Astra-1KR-1L-1M-1N.html), not 12.606 as you wrote and not 12.603 as in your command). The problem is that the NIT Actual is incomplete. It is declared as having 3 sections (numbered 0, 1, 2) but the section 0 is missing. Only sections 1 and 2 are present and repeated. The NIT Other is declared to have only one section (numbered 0) which is present.

By default, the command `tstables` reports each complete table exactly once (per version). Since the NIT Actual is never complete, it is never reported.

You can see that using the following command which reports all individual sections:
~~~
tstables ASTRA_SI.ts --pid 16 --all-sections
~~~

I have tried to increase the capture time to 120 seconds just in case the section 0 would be incorrectly repeated but without luck.

So, as strange as it seems, there is probably an error in the Astra transmission on this specific transponder.

bobben1no, 2019-03-11:
First of all I want to give credit for your work on tsduck.  It is a very useful set of tools.

I have looked at the Astra SI (12.604H) with several analysers and all list all three NIT actual sections (0,1,2).  I also looked at ts packets with hex viewer and it seems like all three NIT sections is transmitted.  The one thing that seem a bit strange is how the NIT sections is (not) aligned in the TS packets.  Seems strange the way the sections is aligned but nothing illegal I assume.  All pointer_fields point at the  position for the table_id byte as far as I can see.

 I can also see stuffing sections with table_id 0x72 on PID 16.  

Here is tree structure from my homemade table decoder:
https://www.satellite-calculations.com/SI/get_tabbed_tree.php?ASTRA_SI.txt

And here is ts packets for complete NIT actual in the order it is received:
```
#16
// start NIT actual section 0 of 2
47 40 10 1F 5D FF FF FF FF FF FF FF FF FF FF FF   G@  ]ÿÿÿÿÿÿÿÿÿÿÿ
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF   ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF   ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF   ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF   ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF   ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ
FF FF 40 F3 EF 00 01 D5 00 02 F0 78 40 08 41 53   ÿÿ@óï  Õ  ðx@ AS
54 52 41 20 31 00 4A 07 04 54 00 01 00 0C 01 4A   TRA 1 J  T     J
07 04 57 00 01 00 0D 01 4A 07 04 57 00 01 00 00     W     J  W    
04 4A 09 04 54 00 01 00 00 04 4E 4F 4A 08 04 57    J  T     NOJ  W
00 01 00 00 0A 02 5F 04 00 00 00 01 4A 07 04 57         _     J  W
00 01 1E E6 90 4A 07 03 EF 00 01 13                  æ J  ï   
#17
47 00 10 10 37 91 4A 07 04 57 00 01 1E E7 93 4A   G   7 J  W   ç J
07 04 06 00 01 11 F7 A2 4A 12 04 06 00 01 11 EE         ÷ J      î
09 0A 00 07 0B 06 00 26 00 01 80 1D F3 6A 03 EB          &  € ój ë
00 01 F0 0D 43 0B 01 12 43 75 01 92 81 02 20 00     ð C   Cu      
04 03 EC 00 01 F0 0D 43 0B 01 12 58 50 01 92 A6     ì  ð C   XP   
02 20 00 02 03 ED 00 01 F0 0D 43 0B 01 12 73 25        í  ð C   s%
01 92 86 02 20 00 02 03 EE 00 01 F0 0D 43 0B 01           î  ð C  
12 88 00 01 92 B6 02 20 00 02 03 EF 00 01 F0 0D              ï  ð 
43 0B 01 13 02 75 01 92 86 02 20 00 02 03 F0 00   C    u        ð 
01 F0 0D 43 0B 01 13 17 50 01 92 A1 02 20 00 04    ð C    P       
03 F2 00 01 F0 0D 43 0B 01 13 47 00 01 92 A6 02    ò  ð C   G     
20 00 02 03 F3 00 01 F0 0D 43 0B 01                   ó  ð C  
#18
47 00 10 11 13 61 75 01 92 86 02 20 00 02 03 F4   G    au        ô
00 01 F0 0D 43 0B 01 13 76 50 01 92 A6 02 20 00     ð C   vP      
02 03 F5 00 01 F0 0D 43 0B 01 13 91 25 01 92 81     õ  ð C    %   
02 20 00 04 03 F7 00 01 F0 0D 43 0B 01 14 20 75        ÷  ð C    u
01 92 81 02 20 00 04 03 F8 00 01 F0 0D 43 0B 01           ø  ð C  
14 35 50 01 92 A6 02 20 00 02 03 F9 00 01 F0 0D    5P        ù  ð 
43 0B 01 14 64 25 01 92 86 02 20 00 02 03 FB 00   C   d%        û 
01 F0 0D 43 0B 01 14 93 75 01 92 86 02 20 00 02    ð C    u       
03 FC 00 01 F0 0D 43 0B 01 15 08 50 01 92 A1 02    ü  ð C    P    
20 00 04 03 FE 00 01 F0 0D 43 0B 01 15 38 00 01       þ  ð C   8  
92 A1 02 20 00 04 03 FF 00 01 F0 0D 43 0B 01 15          ÿ  ð C   
52 75 01 92 86 02 20 00 02 04 01 00               Ru          
#19
47 00 10 12 01 F0 0D 43 0B 01 15 82 25 01 92 86   G    ð C    %   
02 20 00 02 04 02 00 01 F0 0D 43 0B 01 15 97 00           ð C     
01 92 A1 02 20 00 04 04 03 00 01 F0 0D 43 0B 01              ð C  
16 11 75 01 92 81 02 20 00 04 04 04 00 01 F0 0D     u           ð 
43 0B 01 16 26 50 01 92 A1 02 20 00 04 04 07 00   C   &P          
01 F0 0D 43 0B 01 16 70 75 01 92 86 02 20 00 02    ð C   pu       
04 08 00 01 F0 0D 43 0B 01 16 85 50 01 92 A1 02       ð C    P    
20 00 04 04 09 00 01 F0 0D 43 0B 01 09 64 25 01          ð C   d% 
92 86 02 20 00 02 04 0A 00 01 F0 0D 43 0B 01 09             ð C   
79 00 01 92 A1 02 20 00 04 04 0E 00 01 F0 0D 43   y            ð C
0B 01 10 38 00 01 92 A1 02 20 00 04 04 0F 00 01      8            
F0 0D 43 0B 01 10 52 75 01 92 86 02               ð C   Ru    
#20
47 00 10 13 20 00 02 04 10 00 01 F0 0D 43 0B 01   G          ð C  
10 67 50 01 92 A1 02 20 00 04 04 11 00 01 F0 0D    gP           ð 
43 0B 01 10 82 25 01 92 96 02 20 00 03 04 12 00   C    %          
01 F0 0D 43 0B 01 10 97 00 01 92 A1 02 20 00 04    ð C            
04 14 00 01 F0 0D 43 0B 01 11 26 50 01 92 A6 02       ð C   &P    
[... 103 lines cut ...]
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF   ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF   ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF   ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF   ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF   ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF   ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF   ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ
FF FF FF FF FF FF FF FF FF 40 F0 20 00 01 D5 02   ÿÿÿÿÿÿÿÿÿ@ð   Õ 
02 F0 00 F0 13 04 06 00 01 F0 0D 43 0B 01 16 56    ð ð     ð C   V
00 01 92 A6 02 20 00 02 A5 8C DC 6E                         Ün
#29
// start stuffing_section with table_id 0x72:
47 40 10 1C 00 72 F0 B4 FF FF FF FF FF FF FF FF   G@   rð ÿÿÿÿÿÿÿÿ
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF   ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF   ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF   ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF   ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF   ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF   ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF   ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF   ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF   ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF   ÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿÿ
FF FF FF FF FF FF FF FF FF FF FF FF               ÿÿÿÿÿÿÿÿÿÿÿÿ
```

lelegard, 2019-03-14:
Hi,

OK, after in-depth analysis, here is my conclusion. The way the sections are muxed into PID 16 of this stream is at least unusual and probably illegal according to the ISO 13818-1 standard.

If you interpret the standard in a strict way, every two sections, the first section is correctly demuxed and the second one is dropped. By accident, there is always an even number of sections between two instances of the section 0 of the NIT Actual. So, _all_ instances of this section are dropped. As a consequence, the NIT Actual can never be rebuilt since its section 0 is always missing.

If you have a relaxed interpretation of the standard, then you may recover from the incorrect format and extract all sections. I have implemented and pushed such a relaxed implementation and the NIT Actual is now fuly extracted.

Now, if you are interested in the gory details, let's dive into it. Here is a dump of two adjacent packets in PID 16, using `tsdump`:
~~~
* Packet 4
  ---- TS Header ----
  PID: 16 (0x0010), header size: 4, sync: 0x47
  Error: 0, unit start: 1, priority: 0
  Scrambling: 0, continuity counter: 10
  Adaptation field: no (0 bytes), payload: yes (184 bytes)
  ---- Full TS Packet Content ----
  0000:  47 40 10 1A 00 72 F0 B4 FF FF FF FF FF FF FF FF  G@...r..........
  0010:  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ................
  0020:  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ................
  0030:  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ................
  0040:  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ................
  0050:  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ................
  0060:  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ................
  0070:  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ................
  0080:  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ................
  0090:  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ................
  00A0:  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ................
  00B0:  FF FF FF FF FF FF FF FF FF FF FF FF              ............

* Packet 5
  ---- TS Header ----
  PID: 16 (0x0010), header size: 4, sync: 0x47
  Error: 0, unit start: 1, priority: 0
  Scrambling: 0, continuity counter: 11
  Adaptation field: no (0 bytes), payload: yes (184 bytes)
  ---- Full TS Packet Content ----
  0000:  47 40 10 1B 5D FF FF FF FF FF FF FF FF FF FF FF  G@..]...........
  0010:  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ................
  0020:  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ................
  0030:  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ................
  0040:  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ................
  0050:  FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF  ................
  0060:  FF FF 40 F3 EF 00 01 D5 00 02 F0 78 40 08 41 53  ..@........x@.AS
  0070:  54 52 41 20 31 00 4A 07 04 54 00 01 00 0C 01 4A  TRA 1.J..T.....J
  0080:  07 04 57 00 01 00 0D 01 4A 07 04 57 00 01 00 00  ..W.....J..W....
  0090:  04 4A 09 04 54 00 01 00 00 04 4E 4F 4A 08 04 57  .J..T.....NOJ..W
  00A0:  00 01 00 00 0A 02 5F 04 00 00 00 01 4A 07 04 57  ......_.....J..W
  00B0:  00 01 1E E6 90 4A 07 03 EF 00 01 13              .....J......
~~~

The first packet contains a Stuffing Section (rarely used, never seen one in practice, WTF...). The second packet contains the start of the section 0 of the NIT Actual (`40 F3 EF 00 01 D5 00 02` ...)

The problem is not the non-zero pointer field. This is quite usual in highly packed section streams such as EMM's. The problem is all these FF's _before_ the pointed area.

According to ISO 13818-1:

1. In packets with PUSI set, the pointer field points to the start of the first section in this packet.
2. In the absence of missing packets or other out-of-sync, all payloads of all TS packets form a continuous stream of sections. A demux locates the next section based on the length of the previous one.
3. In the specific case a section starts with FF (reserved table id), then the rest of the TS packet shall be ignored. In that case, the next section starts in the next packet where the PUSI is set and the pointer field is zero.

As a consequence, when the pointer field is not zero in a PUSI packet, all bytes before the pointed first section form the end of a previous section which started in a previous packet.

Here, we start the analysis in the first packet. We jump using the pointer field and get the stuffing section. But this section ends exactly at the end of the packet. The next section consequently starts at the next byte in the stream, which is the first byte of the next TS payload, which is FF. According to rule 3., we drop the rest of this packet, dropping the start of section 0 of the NIT Actual. All subsequent packets are also dropped because they have no PUSI and we have no start of section. The resync occurs at the next packet with a PUSI bit. This explains why one section every two sections is dropped.

I modified the code so that an FF starting a section does not drop the packet when the TS packet has PUSI set and the FF triggering the packet drop is _before_ the byte pointed to by the pointer field is after the FF. In that case, we artificially resync inside the packet, at the pointed byte. I do not think that this hurts valid streams and it is a workaround for that kind of stream.

bobben1no, 2019-03-15:
Thank you for your investigation,analysis and workaround.  I got too much focus on the pointer fields and did not see the logical error in the stuffing sections.

closed 2019-03-16 by lelegard