tsduck/tsduck #241: How to scan a DVB-S network (Astra 28.2E UK Freesat)

oviano, 2019-03-14:
Not sure if I'm missing how to do this, but I can capture a stream fine with this command...

tsp --verbose -I dvb --device-name "PCTV 292e/461e BDA 28179 TVTuner S" --frequency 10847000000 --polarity "vertical" --symbol-rate 23000 -P analyze --service-analysis -P until --seconds 10 -O drop > c:\Users\Oliver\Desktop\astra_10847000000_vertical.txt 

...but a scan fails:

tsscan --verbose --device-name "PCTV 292e/461e BDA 28179 TVTuner S" --frequency 10847000000 --polarity "vertical" --symbol-rate 23000
tsscan: cannot scan network, no NIT found on specified transponder 

Also, am I correct in assuming it's not possible to do a full scan off DVB-S? When I leave out the tuning parameters above, it tries to scan using DVB-T frequencies:

tsscan --verbose --device-name "PCTV 292e/461e BDA 28179 TVTuner S"
scanning channel 21, 474,000,000 Hz
scanning channel 22, 482,000,000 Hz
scanning channel 23, 490,000,000 Hz
scanning channel 24, 498,000,000 Hz
scanning channel 25, 506,000,000 Hz
scanning channel 26, 514,000,000 Hz
scanning channel 27, 522,000,000 Hz
scanning channel 28, 530,000,000 Hz
scanning channel 29, 538,000,000 Hz
scanning channel 30, 546,000,000 Hz
scanning channel 31, 554,000,000 Hz
scanning channel 32, 562,000,000 Hz
scanning channel 33, 570,000,000 Hz
scanning channel 34, 578,000,000 Hz
scanning channel 35, 586,000,000 Hz
scanning channel 36, 594,000,000 Hz
scanning channel 37, 602,000,000 Hz
scanning channel 38, 610,000,000 Hz
scanning channel 39, 618,000,000 Hz
scanning channel 40, 626,000,000 Hz
scanning channel 41, 634,000,000 Hz
scanning channel 42, 642,000,000 Hz 
etc

WinTV has an option where you choose the satellite, so I'm guessing for Tsduck to support DVB-S scanning it would require some sort of file detailing all the satellites and their frequency ranges? Maybe there is some kind of online resource it could use?

lelegard, 2019-03-14:
There is no single definition for the word "scan". `tsscan` implements two methods: UHF/VHF band or NIT scan.

1. The UVH/VHF band scan is typically for terrestrial networks. This is the default. Since the band is limited to typically max 50 different known frequencies, the scan operation tests them all. Moreover, knowing the exact tuning parameters is not necessary to a successful tuning.

2. When tuning parameters are specified, `tsscan` reads the NIT from that transponder. The NIT is then analyzed and all referenced transponders - with all tuning parameters -  are found here. This is the typical scanning method for a sat operator's STB.

In your case, you tried a NIT scan but the transponder contained no NIT (see message). No NIT in a sat transponder means no way to scan the network from this transponder. You need to use the "root transponder" (if any) from the operator.

In the last command, you did not specify tuning parameters. So, `tsscan` switched to a UHF band scan, which is a non-sense on a satellite.

If a sat operator has several transponders but no NIT, you need to know the transponders' parameters from the beginning.

oviano, 2019-03-14:
Perfect that explains everything, thank you.

Turns out the Freesat "home transponder" is

closed 2019-03-16 by lelegard