tsduck/tsduck #242: ATSC scanning issues

oviano, 2019-03-14:
Firstly, I know this code was only recently submitted so maybe it's not complete, but I've got access to the machine with ATSC reception again and I was trying to scan to produce the channels XML file.

I'm finding that I can manually tune and capture the transport stream if I specify the frequency. The code we discussed and that I added recently to map frequency -> physicalChannel appears to work. So that's good, as it's the first time I've been able to test it.

However, running a scan, it goes through and lists the correct physical channels and frequencies, and reports it has a locked signal for many of them (including the one I was able to manually capture above), but never lists any transport stream IDs.

I guess I'm going to need to step through and debug this, unless you can think of anything obvious, or as I mention above, maybe I'm trying this code too soon.

lelegard, 2019-03-14:
Hum, it should work "in theory". But, without ATSC network in the area (France), I can only test captured streams while `tsscan` needs a tuner and a network by definition. I am afraid you will have to debug. Or maybe simply start with `tsscan --debug --service-list`.

oviano, 2019-03-14:
I will set myself the task to figure it out :)

I’ll report back with my findings...

lelegard, 2019-03-14:
Thanks. You did a good job on ATSC support with DirectShow. I am sure you will find the solution.

oviano, 2019-03-15:
Well you can close this, as it was abject stupidity on my part. I had made some changes to my local code to try and get some sensible signal strength values on Windows (it always shows 100%). This worked for DVB-T but screwed up tuning on ATSC.

The good news is ATSC tuning works, and my code (lifted from the -P zap) works without modification.

closed 2019-03-15 by oviano