tsduck/tsduck #243: Feature Request: store and search in channels XML file for major/minor channel (ATSC)

oviano, 2019-03-16:
Without having looked at the code I could probably look at doing this myself if it doesn’t appeal to you, @lelegard.

lelegard, 2019-03-18:
Done. The XML format of the channels configuration file has been enhanced to include major and minor id of ATSC channels. This means that you need to `tsscan` again to rebuild the channel configuration with the new information.

The option `--channel-transponder` can now use "major.minor", eg. `--channel-transponder KXYZ` or `--channel-transponder 1.4`.

Please tell us if this works on a real ATSC network.

lelegard, 2019-03-18:
Yes, gathering various streams is always useful. Each operator or country has its own percularities which are useful to know and test.

oviano, 2019-03-20:
I can confirm the scan correctly adds the major/minor channels to the XML file, and that channel-transponder correctly tunes to the right frequency.

A couple of observations though:

-P zap "service_name" doesn't seem to work for ATSC. I haven't tried this until now, as my own code uses the XML file searching option to get the service ID for a service name and then does the equivalent of zap on the service id. Is this intended to be working now, or is it not yet implemented?

- Logically you could also allow -P zap "major.minor" as another way of specifying it?

lelegard, 2019-03-21:
I just added the zap by name or by major.minor for ATSC streams.

There is one restriction though: The xVCT is not modified in the resulting SPTS. Technically, the output is a clean SPTS with only one service in the PAT. But the ATSC xVCT still contains names of services which were removed. On DVB streams, the SDT is cleaned up to contain only the description of the remaining service.

The reason for this restriction is that, on ATSC streams, the PID 0x1FFB is a mixture of cycled tables (MGT, xVCT) and one-shot tables (STT). It is consequently more difficult to modify tables in this PID. On DVB streams, the PID 0x0011 contains cycled tables only (SDT, BAT).

In fact, the TSDuck packetizer (the section remux) was designed to work either with cycled / scheduled sections or one-shot sections but not a mixture of the two (the design must be completely different). This works fine for MPEG base and DVB. But ATSC introduces what I consider a poor signalization design in the PSIP PID.

closed 2019-04-14 by lelegard