tsduck/tsduck #246: ATSC/Windows. Glitch at start of audio streams on one frequency only.

oviano, 2019-03-19:
I don't know if this is a tsduck issue, or a Windows driver issue, but there is one frequency that exhibits a strange problem. It happens every time, so it's not reception/lap of the Gods/day of the week-related.

The first audio packets on both streams fail with an invalid data error when I try and pass them to FFmpeg libraries. So presumably the packet is messed up in some way, although it appears to have a valid pts/dts, duration, size etc.

I can survive that, by ignoring it. The second packet then seems ok, producing no errors inside the FFmpeg libraries when asked to decode it.

Then the third packet arrives, but with the same dts/pts as the second packet. If I ignore the third packet, everything decodes peacefully from then on.

So I can work around it, either by specific code to track the above situation, or something more generic like FFmpeg itself does in such cases, which seems to be to keep the third packet but shift all the timestamps on thereafter.

But it strikes me as rather odd that this happens for both audio streams, and only on this particular frequency. It can't be the transmission (I assume?) because the errors are only at the start of the stream, and for a live transmission there isn't a concept of the "start of the stream", it's just continuous data which we tune into and start receiving at some random point in the middle of it.

lenosj, 2019-03-19:
Do any of the packets in the first and third audio packets have the
continuity or error bits set?
Is it possible that these packets were adjacent to one another before
re-interleave and it's just an artifact of jumping into the middle of a
stream.
Being the third audio packet, this is a long-shot but I don't know the
offset into the stream for the third packets.

oviano, 2019-03-19:
I just checked, and neither the transport error indicator or discontinuity indicator bits are set for any of the packets received.

lelegard, 2019-03-19:
Weird or missing packets at the beginning of a capture on Windows is something I have experienced from day 1 (meaning about 10 years ago). I would bet it is a DirectShow issue rather that a driver one since it occurs for me with various brands of tuners, on DVB-S and DVB-T (and ATSC for you). 

May I suggest (again) to use Linux 😉 I have never experienced that on Linux.

In verbose mode, do you see messages such as "tuner packet synchronization lost, dropped NN bytes in NN chunks" ? In that case, the TSDuck `SinkFilter` identified explicitly incorrect data stream coming from DirectShow and tries to recover. But the packet recovery from an incorrect data stream may not always be correct.

As a rule of thumb, on Windows, I recommend to drop a few initial packets. On Windows, I usually use commands like `tsp -I dvb ... -P skip 100 ...`. If this is an application you write with the TSDuck library, simple do not use the first N packets (100 is just an example but a reasonable one).

oviano, 2019-03-19:
No I don’t see those sync errors. I’ve seen those before (as per my other thread where I was trying to investigate it) but not in this case.

Yes this is my streaming server application linking to the Tsduck libs.

Thanks for the tip re just skipping a bunch of packets early on.

closed 2019-03-21 by lelegard