tsduck/tsduck #247: small issue in website - link doesn't work

nautilus7, 2019-03-19:
Hi, I am trying to download the build tools needed to [compile tsduck for windows](https://tsduck.io/doxy/building.html). The download link for [Graphviz for Windows](http://www.graphviz.org/Download_windows.php) doesn't work. It should be replaced with something like https://graphviz.gitlab.io/_pages/Download/Download_windows.html for example.

Anyway, I would also like to ask when newer binaries will be available, because I would like to use some of the latest features. If new binaries will be available soon, perhaps you could save me the trouble of building my own.

Thanks a lot.

lelegard, 2019-03-19:
Hi @nautilus7,

Thank you for reporting this. They must have moved their web site. I just fixed the link.

There are a few things to complete before releasing version 3.17. I have just started a rebuild of the installers on a Windows VM on my Mac. Since Visual Studio is so slow to build, I think we will have donner and then I will publish the (non final) binaries in the test area. I will post links here. Stay tuned...

lelegard, 2019-03-19:
Installers for intermediate version 3.17-1177 are available here:

- [TSDuck-Win64-3.17-1177.exe](https://tsduck.io/download/test/TSDuck-Win64-3.17-1177.exe)
- [TSDuck-Win32-3.17-1177.exe](https://tsduck.io/download/test/TSDuck-Win32-3.17-1177.exe)
- [TSDuck-Win64-3.17-1177-Portable.zip](https://tsduck.io/download/test/TSDuck-Win64-3.17-1177-Portable.zip)
- [TSDuck-Win32-3.17-1177-Portable.zip](https://tsduck.io/download/test/TSDuck-Win32-3.17-1177-Portable.zip)

I hope it helps...

nautilus7, 2019-03-19:
I am very very thankful for the binaries!

closed 2019-03-19 by nautilus7