tsduck/tsduck #248: scrambler plugin does not add CA_descriptor in PMT

nautilus7, 2019-03-20:
Hi, I am trying to scramble a ts file using a fixed cw and the scrambler plugin. The service id is 11110. The command line I use is this:

`tsp -I file full.ts -P scrambler --pid-ecm 8191 --cw 1122334455667788 --dvb-csa2 11110 -O file encrypted.ts`

The resulting ts file does not contain any CA descriptors in the PMT. What am I missing? Thanks.

lelegard, 2019-03-20:
Hi @nautilus7 

When you specify a fixed CW, you just do scrambling but you are not under the control of a Condition Access System. So, without CA, there is no CA descriptor.

A conditional access system distributes access rights in EMM's and specifies access criteria in ECM's. To use a CA system during scrambling, you need:

1. Let the `scrambler` plugin generate random CW by crypto-period.
2. Connect to a real ECMG system, as provided by the CA you want to use (option `--ecmg`).

If you just want to experiment, you may use `tsecmg`, an ECMG emulator from TSDuck. But this is a fake CAS, there is no security in the pseudo-ECM's.

You can put it in another way: A CA descriptor in a PMT specifies an ECM PID. ECM's are generated by an ECMG. No ECMG => no ECM => no ECM PID => no CA descriptor.

nautilus7, 2019-03-20:
Hi, I am trying to experiment with BISS1 and BISS2  mode 1 encoding. They both use fixed control word, so there is no need for actual ECM or ECM pid. Just a CA_descriptor in the PMT with fake ecm pid 0x1FFF and ca_system_id 0x2600 or 0x2602 respectively.

I managed to add a PMT descriptor using the pmt processor plugin for tsp. It can do exactly what I want.

My initial confusion was created by these lines in the scrambler plugin manual:

```
--component-level
Add CA_descriptors at component level in the PMT. By default, one CA_descriptor is added at
program level.
```
I thought that a CA_descriptor will be added anyway. But I guess this is done only when using the ecmg plugin, right?

@lelegard, I also found a couple of typos in the manual. I report them here to avoid opening a new issue:

Section 3.1.2 page 20: _When an option or parameter is documented **to an require integer value** (PID, identifier, etc.), this value_

Page 119 - `duplicate` tsp plugin: _This plugin duplicates the content of several PID’s into **news** PID’s. The duplicated packets are created by_

lelegard, 2019-03-20:
Hi,

Thanks for reporting typos, I just fixed them. I also improved the documentation of the scrambler plugin to specify that using fixed control words completely bypasses all CAS features (ECM, ECMG, CA descriptors).

Beyond accidental typos such as those two, I would also appreciate editorial reviews from native English speakers. Being French, I cannot guarantee that the documentation is written in good English. So, whether you are a brexiter or a yankee (wallabies also accepted 😄 ), if you find that some sentences is incorrect, awkward or whatever, please report this and suggest improvement. Thanks  in advance.

nautilus7, 2019-03-20:
No, I am not a native English speaker, but I'll have it in mind to check for such things whenever I read the docs.

Thanks for this great application!

closed 2019-03-20 by nautilus7