tsduck/tsduck #249: Scrambler a multicast stream

AndersonDan, 2019-03-20:
How do I do the scrambler of a TV stream, for example, coming from the multicast udp://@241.0.0.198:1112 (an example only) and output by the address udp://@241.0.1.198:8787?

I tried the command below, but to no avail. The flow works (wheel in VLC) but it is not with scrambler.

    tsp -I ip 241.0.0.198:1112 -P scrambler --cw 1122334455667788 --dvb-csa2 11110 -O ip 241.0.1.198:8787

lelegard, 2019-03-20:
> I tried the command below, but to no avail

What do you mean precisely? Did you analyze the produced stream? 

All plugins are independent and can be replaced without affecting other plugins. When analyzing a stream, replace the output plugin with `-O file ...` and quietly analyze the file. It contains exactly the same thing that would be sent by `-O ip`.

AndersonDan, 2019-03-21:
> What do you mean precisely? Did you analyze the produced stream?

Theoretically, the stream should not run in VLC without informing the same key to descrambler. It turns out that in VLC runs like a scrambler, it's as if the plugin was actually ignored.

lelegard, 2019-03-21:
That is why I advised to save the output of tsp in a file. Then, you can inspect the output of tsp, see if packets are actually scrambled, etc. You cannot troubleshoot a problem by simply observing the end-to-end system. You need to split the various steps and analyze what is exchanged in between. This is exactly what I had in mind when designing TSDuck.

AndersonDan, 2019-03-22:
I found my mistake. If I do not set the PIDs correctly the scrambler does not work correctly. Defining, it works.
Is there any way for TSDuck to automatically identify PIDs?
Another question is: is it possible to set even and odd keys in the scrambler?

lelegard, 2019-03-22:
> Is there any way for TSDuck to automatically identify PIDs?

What do you mean by "identify PIDs"? Identify which PID? If you specify a service name or id, the plugin will scramble all video and audio PID's for this service, as declared in the PMT of the service. Various options alter the selection of which PID to scramble in the service. Alternatively, instead of a service, you may explicitly specify a list of PID's to scramble, regardless of any service structure.

> Another question is: is it possible to set even and odd keys in the scrambler?

If you specify a list of fixed control words (option `--cw-file`), each CW is used as even, odd, even, odd, etc. The trigger between even / odd is the crypto-period duration.  

But using fixed CW is reserved to very specify unitary tests. In the real world, scrambling in used in cooperation with a Conditional Access System. The CW are randomly selected for each crypto-period and wrapped into ECM's, using an ECMG from the specified CAS to generate the ECM's.

closed 2019-03-23 by AndersonDan