tsduck/tsduck #250: HLS manifest with SCTE-35

Kirill-Shkodkin, 2019-03-20:
Hi, TSDuck team!

Your tool and doc is pretty cool! Thank for it )

I have a question: is it possible to generate hls(output) with CUE-IN, CUE-OUT splices according to HLS doc? I see SCTE-35 information in generated .ts files, but not in playlist.

I use this command:
`tsp -P zap 0x0001 -P spliceinject --pid 0x0029 --pts-pid 0x0029 --udp 1000 -I ip 10.10.10.2@238.1.1.1:1000 -O hls -d 4 -p hls.playlist.m3u8 stream.ts `

jacksonvm, 2019-03-20:
Hi @lelegard ,

This is option will be really good if we can decorate the manifest with the CUE-IN and CUE-OUT. I have slight confusion in segmentation of the TS if the SCTE inserted is on a PTS which is not part of an I FRAME. 

Regards,

Jackson

lelegard, 2019-03-20:
No, the HLS CUE-* options are not implemented. This does not fit very well with the architecture of tsp. The generation of HLS manifests and segments is the job of an output plugin while the manipulation of the stream is dedicated to packet processing plugins.

mhzbg, 2019-05-31:
Correct me if I am wrong, but there is already an HLS output plugin. This request will fall under this plugin. Decorating the manifests if there are SCTE-35 markers in the stream will simply need to be reflected in the manifests at the right times if source contains them. Now this is the easy part. The hard part is that the segments need to be cut on such, and the issue here is that TSDuck does not do any encoding, so it wont be able to split the TS frame accurate on the marker, which is the requirement for these markers in first place. I am not sure if manifest decoration will be much useful without this. The real world use case with these are AD insertion by SSAI services such as AWS MediaTailor and Google DFP, which all utilise the CUE markers and replace them with ad segments. Elemental encoders do this pretty well.

zozzancs, 2019-07-31:
Anyone found a workaround for this? tsp fork to ffmpeg and create hls with that maybe?

fraroas, 2022-12-16:
https://github.com/futzu/x9k3 will do your job