tsduck/tsduck #251: datainject questions

adambever1, 2019-03-21:
Hello,

How are you able to use tsemmg with datainject as described in the documentation, "This fake EMMG can be used with the tsp plugin named datainject to build an end-to-end demo of a DVB SimulCrypt system."

Do you have an example of this?

Best Regards,

Adam

lelegard, 2019-03-21:
In the TSDuck user's guide, section 5.2.19 _"Complete Conditional Access System test bed"_, there is an example of a `tsp` command with all features of a CAS, including interfacing with an EMMG using the plugin `datainject`. Now, simply run `tsemmg` from another session or another computer, using the TCP server port you specify in `datainject`. All parameters for `tsemmg` are in user's guide as well.

The EMMG (here `tsemmg`) connects to the MUX (here the `datainject`plugin in the `tsp` command) and injects its EMM's (fake data in the case of `tsemmg`).

adambever1, 2019-03-21:
```
tsemmg -m localhost:25000
tsemmg: error connecting socket: Connection refused
```
Running the command above gets me the error above. So my understanding of that is that it is unable to connect to the MUX (here being datainject). However it seems like a chicken or the egg scenario, which comes first?

lelegard, 2019-03-22:
With this message, I see two possibilities at least:

1. You have a networking issue, firewall, iptables, etc.
2. The `datainject` plugin does not listen on port 25000. Either because the `tsp` process terminates too early or it uses another port.

Note that this is "simply" a TCP issue, so just focus on TCP, not on DVB SimulCrypt issues.

Note that data injection is mostly a real-time process. If your input is a file, then you should play it at its playout rate, not at the disk reading speed. Use plugin `regulate` if necessary (see examples in the user's guide).

adambever1, 2019-03-27:
Hi Thierry,

Thanks for the response it has been very helpful. The regulate command did the trick!

Best Regards,

closed 2019-03-27 by adambever1