tsduck/tsduck #252: HLS playing problems

joolzg, 2019-03-23:
Running live i am seeing problems with VLC.

The command line is the standard '-O hls -l 5 -d 12 --playlist 1000.m3u8 1000-.ts'

Now the problem is the m3u8 contains the 5 segments, but it contains the segment being created.

so
#EXTINF:11.90
1000-000000.ts
#EXTINF:11.90
1000-000001.ts
#EXTINF:11.90
1000-000002.ts
#EXTINF:11.90
1000-000003.ts
#EXTINF:11.90
1000-000004.ts

Files sizes
1501231 1000-000000.ts
1563232 1000-000001.ts
1812342 1000-000002.ts
1672623 1000-000003.ts
    89000 1000-000004.ts

So something be done to only use "complete" segments

lelegard, 2019-03-25:
Hi @joolzg 

I cannot make tests this week since I am traveling. However, looking at the code, it seems that:

- A segment is added to the playlist only after closing it (this is the intended behaviour).
- If a segment contains PCR's, its duration in the playlist (`EXTINF`) is based on the PCR values.
- If there is no or not enough PCR's in a segment, its duration in the playlist (`EXTINF`) is always set to the target duration.

See [here](https://github.com/tsduck/tsduck/blob/master/src/libtsduck/tshlsOutputPlugin.cpp#L272)

In your case, based on the ratio between file sizes, the duration of the last segment is approximately 0.6 second. So, there is probably not enough PCR inside it.

In that case, if there is a previous segment with a known bitrate, I should probably use the previous bitrate to extrapolate the duration of the short segment.

joolzg, 2019-03-25:
When you could look at this, but as you see from my debug output the last segment in the m3u8 is still being written.

lelegard, 2019-03-25:
I do not see debug output. I see the content of the playlist and the file sizes. There is an inconsistency in the fact that the playlist says that segment 4 has the same duration as other segments while its file size is 10 times smaller. From the code (to be confirmed by tests), I would say that

-  the process is complete,
- the last segment is complete and closed,
- its size is incorrect in the playlist

If you have debug log which shows something different, please post it here.

I will test next week-end if necessary.

joolzg, 2019-03-25:
Sorry the times are copied by me and the last is a mistake

I can try running again and see of i can replicate.

Question, do you write the M3U8 before the TS?

closed 2019-03-25 by joolzg