tsduck/tsduck #253: HLS Initial segment number

joolzg, 2019-03-24:
When restarting it would be good to be able to specify the start number for the segment, so instead of going back to 0 we could restart from an arbitary point

--start-segment-number X

lelegard, 2019-03-25:
Hi @joolzg, The option `--start-segment` has been added in version 3.17 as a response to request #197 from ... @joolzg 😉

joolzg, 2019-03-25:
This is for HLS output, sorry

lelegard, 2019-03-25:
What do you mean by start segment number in an output playlist?
Is it the value of `#EXT-X-MEDIA-SEQUENCE` ?

joolzg, 2019-03-25:
Yes, as if a stream stops and restarts it would be nice to carry on with the sequence instead of starting at 0

lelegard, 2019-03-26:
Option `--start-media-sequence` has been added.

joolzg, 2019-03-30:
Thanks again

closed 2019-03-30 by joolzg