tsduck/tsduck #255: SDT creation

anr27, 2019-04-01:
Hello.
First of all thank you for this great toolkit ! It is a really nicely put set of instruments and quite extensive documentation.
The issue I have is with SDT creation - I am trying to add an SDT to an incoming HLS stream without one and output it to UDP multicast.
I have added stuffing with TSP options and tried using SDT plugin options (-c , --create-after, -bitrate and --inter-packet ) in various combinations however SDT is not being created. The debug message I receive is : "Debug: sdt: will create SDT after 0 packets, 100 ms (bitrate: 0 b/s)".

lelegard, 2019-04-01:
Hi,

If there is an existing SDT, its PID is replaced. If there is none, its PID needs to be created and new packets must be inserted.

The usual reason for not having the expected data inserted if the lack of stuffing. The `tsp` command cannot  add new packets in a stream. It can only modify or delete existing packets. A broadcast stream usually contains stuffing that can be replaced with new packets. HLS streams have usually no stuffing and it is impossible to insert anything. There are techniques such as adding artificial stuffing at the input level (option `--add-input-stuffing`) which create opportunities for packet insertion. See the documentation of the `tsp` command in the TSDuck user's guide, especially the section named "Modifying, inserting and deleting packets"). 

Note that if you add input stuffing to be replaced later by new packets (here, SDT), the new packets will not use them all and a lot of stuffing will remain on output. It is recommended to add a trailing `filter` plugin at the end of the chain to remove the extra stuffing.

anr27, 2019-04-01:
Hi, thank you for this fast feedback.
I have read the documentation (particuand added stuffing as described but SDT is still not created.

lelegard, 2019-04-01:
OK, could you please post the complete command and the corresponding log messages (`tsp --verbose`)?

anr27, 2019-04-02:
/usr/bin/tsp -d3 -a 1/2 -t -v -I hls --highest-bitrate http://<link> -P remap 4095=5023 256=1023 257=2023 -P pmt -i 23 -P pat -a 23/5023 -r 1 -P sdt --create-after 100  --inter-packet 100  -s 23 --original-network-id 1663 -n "Network" --ts-id 23 -p "Network" -P pcrbitrate --min-pcr 256 -P regulate --packet-burst 128 --pcr-synchronous -O ip --enforce-burst 239.16.63.23:30000

* 2019/04/02 09:55:49 - hls: downloading from http://<link>.ts
* 2019/04/02 09:55:49 - hls: MIME type: unknown, expected size: unknown
* 2019/04/02 09:55:49 - Debug: hls: HTTP header: HTTP/1.1 200 OK
* 2019/04/02 09:55:49 - Debug: hls: HTTP header: Accept-Ranges: bytes
* 2019/04/02 09:55:49 - Debug: hls: HTTP header: Access-Control-Expose-Headers: Date, Server, Content-Type, Content-Length
* 2019/04/02 09:55:49 - Debug: hls: HTTP header: Server: WowzaStreamingEngine/4.5.0
* 2019/04/02 09:55:49 - Debug: hls: HTTP header: Cache-Control: no-cache
* 2019/04/02 09:55:49 - Debug: hls: HTTP header: Access-Control-Allow-Origin: *
* 2019/04/02 09:55:49 - Debug: hls: HTTP header: Access-Control-Allow-Credentials: true
* 2019/04/02 09:55:49 - Debug: hls: HTTP header: Access-Control-Allow-Methods: OPTIONS, GET, POST, HEAD
* 2019/04/02 09:55:49 - Debug: hls: HTTP header: Access-Control-Allow-Headers: Content-Type, User-Agent, If-Modified-Since, Cache-Control, Range
* 2019/04/02 09:55:49 - Debug: hls: HTTP header: Date: Tue, 02 Apr 2019 06:55:49 GMT
* 2019/04/02 09:55:49 - Debug: hls: HTTP header: Content-Type: video/MP2T
* 2019/04/02 09:55:49 - Debug: hls: HTTP header: Content-Length: 3448484
* 2019/04/02 09:55:49 - Debug: hls: announced content size: 3448484 bytes
* 2019/04/02 09:55:49 - Debug: sdt: will create SDT after 0 packets, 100 ms (bitrate: 0 b/s)
* 2019/04/02 09:55:49 - Debug: sdt: will create SDT after 0 packets, 100 ms (bitrate: 0 b/s)
* 2019/04/02 09:55:49 - Debug: sdt: will create SDT after 0 packets, 100 ms (bitrate: 0 b/s)
* 2019/04/02 09:55:49 - Debug: sdt: will create SDT after 0 packets, 100 ms (bitrate: 0 b/s)

lelegard, 2019-04-05:
I found the issue. The SDT does not exist and will be created after 100 ms, _in term of broadcast time_. So, tsp must compute how many packets make 100 ms. To do this, the bitrate must be known. Here, it is unknown...

There is no problem when the input bitrate is known from the beginning, ie. in any of the following cases:
- The input is a DVB tuner and the bitrate is precalculated from the tuning parameters.
- The initial preload of the buffer contains enough PCR's to estimate the bitrate.

Here, the initial preload is not enough to estimate the bitrate. 

This is of course a problem. I fixed it and pushed the fix. Please pull and [rebuild](https://tsduck.io/doxy/building.html) if you need this fix.

anr27, 2019-04-05:
Just pulled the source from github and tried to rebuild but complication brakes at this point :
  [CXX] tsDVBCharset.cpp
tsDVBCharset.cpp:36:32: error: redeclaration 'ts::DVBCharset::DVB_SINGLE_BYTE_CRLF' differs in 'constexpr'
 const uint8_t  ts::DVBCharset::DVB_SINGLE_BYTE_CRLF;
                                ^
In file included from tsDVBCharset.cpp:30:0:
tsDVBCharset.h:60:34: error: from previous declaration 'ts::DVBCharset::DVB_SINGLE_BYTE_CRLF'
         static constexpr uint8_t DVB_SINGLE_BYTE_CRLF = 0x8A;
                                  ^
tsDVBCharset.cpp:36:32: error: declaration of 'constexpr const uint8_t ts::DVBCharset::DVB_SINGLE_BYTE_CRLF' outside of class is not definition [-fpermissive]
 const uint8_t  ts::DVBCharset::DVB_SINGLE_BYTE_CRLF;
                                ^
tsDVBCharset.cpp:37:32: error: redeclaration 'ts::DVBCharset::DVB_CODEPOINT_CRLF' differs in 'constexpr'
 const uint16_t ts::DVBCharset::DVB_CODEPOINT_CRLF;
                                ^
In file included from tsDVBCharset.cpp:30:0:
tsDVBCharset.h:65:35: error: from previous declaration 'ts::DVBCharset::DVB_CODEPOINT_CRLF'
         static constexpr uint16_t DVB_CODEPOINT_CRLF = 0xE08A;
                                   ^
tsDVBCharset.cpp:37:32: error: declaration of 'constexpr const uint16_t ts::DVBCharset::DVB_CODEPOINT_CRLF' outside of class is not definition [-fpermissive]
 const uint16_t ts::DVBCharset::DVB_CODEPOINT_CRLF;
                                ^
make[2]: *** [release-x86_64/tsDVBCharset.o] Error 1

I am building on
CentOS Linux release 7.6.1810 (Core)
Kernel 3.10.0-957.10.1.el7.x86_64

lelegard, 2019-04-05:
OK, sorry, my fault. CentOS runs an outdated GCC 4.8.5 compiler. For day-to-day work, I compile on macOS, Fedora and Windows. I normally compile on CentOS before releasing an official version only and there are sometimes a few compilation issues for fix. That is the case today with some code that was modified since the last official TSDuck version. I fixed them. Please pull again and rebuild. Sorry for the inconvenience.

anr27, 2019-04-08:
Pulled, compiled, works as a charm. Thank you once again,  Thierry. Appreciate your work.

closed 2019-04-08 by anr27