tsduck/tsduck #256: Continuity counter should not increment in some cases

alevinsn, 2019-04-03:
According to ISO/IEC 13818-1, Section 2.4.3.3 (as described at https://stackoverflow.com/questions/24310441/mpeg-transport-stream-continuity-counters ), the continuity counter should not be incremented under certain circumstances:
a) Packet has no payload
b) The discontinuity flag is set
c) Packet is a duplicate of the last packet

As far as I can tell, tsduck doesn't do anything special for any of these cases, so if a packet has no payload yet has its CC incremented compared to the last packet of the same PID, tsanalyze doesn't report this as a continuity error.  This threw me for a loop, as tsanalyze reported no continuity errors while some other tools reported continuity errors (VLC and DVBinspector) for a TS file that I had generated using tsduck but for which I had to fix the CCs ala the approach taken in the rmsplice plugin.

Further, there is at least one plugin, the rmsplice plugin, that doesn't adhere to these rules in the fixcc code.  The fixcc code should look more like the following:

		// should not increment CC if no payload or if discontinuity indicator is set
		// also should not do in the case of duplicate packets
		bool bIncrementCC = true;
		if (!pkt.hasPayload())
			bIncrementCC = false;
		else if (pkt.getDiscontinuityIndicator())
			bIncrementCC = false;
		// possible TODO:  detect duplicate packets--should not increment in this case
		// could use VLC's approach of keeping the last packet for each PID and comparing the memory of each
		// however, duplicate packets shouldn't typically come up in practice
		if (bIncrementCC)
			pkt.setCC((ccstate.m_cc + 1) & CC_MASK);
		else
			pkt.setCC(ccstate.m_cc & CC_MASK);

There are other places in the tsduck codebase that call setCC(), and this may be relevant for those as well.

lelegard, 2019-04-03:
Agreed, I will look into it.

I have only one remark about so-called "duplicate" packet. It is sometimes legitimate to have a sequence of two identical but distinct packets. Typically, we see sequences of identical all-FF packets in video streams. 

- On a receiver side, this means that we simply check the CC and drop a packet if its CC is the same as the previous one (either duplicate packet or no payload).
- On a generator side, we never send duplicates. Duplicates can appear in the network.
- The problem is the "fixer" side, as in TSDuck. If we keep a copy of the previous packet and the next one is identical (except the CC), then this is not necessarily a duplicate. We need to "replicate the duplicate" without collapsing sequences of identical packets as false duplicate.

lars18th, 2019-04-03:
Hi,

I, too, have another observation about "no payload" packets. A packet without payload may contain useful information in the header, e.g. PCR marks. Therefore, in this case the CC counter must be increased.

So, please, if you provide any new behavior, include some options to enable or disable this change.

Regards.

alevinsn, 2019-04-03:
> I, too, have another observation about "no payload" packets. A packet without payload may contain useful information in the header, e.g. PCR marks. Therefore, in this case the CC counter must be [...]

Regarding lars's comment, the CC should not be incremented in this case.  In fact, incrementing the CC in this case results in serious problems with playback in VLC, as I personally encountered.  As soon as I stopped incrementing the CC for packets without any payload, the VLC issues went away.  Not all players care about this though.  For example, both Kodi and MPC-HC aren't impacting by incorrect CC increments.

Here are the relevant paragraphs from the actual specification:

"continuity_counter – The continuity_counter is a 4-bit field incrementing with each Transport Stream packet with the same PID. The continuity_counter wraps around to 0 after its maximum value. The continuity_counter shall not be incremented when the adaptation_field_control of the packet equals '00' or '10'.

In Transport Streams, duplicate packets may be sent as two, and only two, consecutive Transport Stream packets of the same PID. The duplicate packets shall have the same continuity_counter value as the original packet and the adaptation_field_control field shall be equal to '01' or '11'. In duplicate packets each byte of the original packet shall be duplicated, with the exception that in the program clock reference fields, if present, a valid value shall be encoded.

The continuity_counter in a particular Transport Stream packet is continuous when it differs by a positive value of one from the continuity_counter value in the previous Transport Stream packet of the same PID, or when either of the nonincrementing conditions (adaptation_field_control set to '00' or '10', or duplicate packets as described above) are met.  The continuity counter may be discontinuous when the discontinuity_indicator is set to '1' (refer to 2.4.3.4). In the case of a null packet the value of the continuity_counter is undefined."

alevinsn, 2019-04-03:
> * The problem is the "fixer" side, as in TSDuck. If we keep a copy of the previous packet and the next one is identical (except the CC), then this is not necessarily a duplicate. We need to "replicate the duplicate" without collapsing sequences of identical packets as false duplicate.

With TSDuck, at least with the rmsplice example that I cited earlier, the plug-in does act as a pseudo-generator.  If, say, the input packets had been duplicates and marked as such by whatever created the input media by not incrementing the CC, this non-increment of the CC will be discarded, since the code increments the CC regardless of the content of the packets.  Perhaps there are other cases of this in the tsduck source base.

lars18th, 2019-04-04:
Hi,

With all due respect, just one comment about my request: I only say to include the *OPTION* to enable or disable the new functionality requested by @alevinsn .

From my point of view the current behaviour is correct (in reference to the CC +1 when no payload). And a lot of STBs will have troubles if you force to not increment the CC when no payload. From my experience the VLC implementation isn't very good in several aspects. The developers doesn't agree to fix some bugs for years, and prefer sometimes to overcome the specs.

I do not doubt to not increase the CC when the error flag is present. But I don't agree to not do it when there is no payload but there is adaptation field (so header info present).

So, I hope @lelegard will consider this an **optional** functionality.

lelegard, 2019-04-05:
Hi @lars18th 

>... include the OPTION to enable or disable the new functionality requested by @alevinsn .
> From my point of view the current behaviour is correct (in reference to the CC +1 when no payload). 

@alevinsn is right and TSDuck is currently wrong. The ISO 13818-1 is very clear about that:

> __continuity_counter__ – The continuity_counter is a 4-bit field incrementing with each Transport Stream packet with the same PID. The continuity_counter wraps around to 0 after its maximum value. [...]

I cannot imagine STB's or VLC having troubles with such thing. 

It is sometimes useful to make invalid behaviour an option to satisfy buggy equipments. And buggy is the most common characteristics of most STB middleware. But the CC are usually handled at hardware demux level, not middleware. So, in this specific case, I do not think that this would be appropriate to enable incorrect behaviour.

lelegard, 2019-04-07:
Hi @alevinsn,

I have (hopefully) fixed this. Now, all CC processing are centralised in one single class, `ContinuityAnalyzer`, which can be used either as an analyser, fixer or generator. All plugins and utility classes now use this one.

Please confirm that the new behaviour is ok for you.

lelegard, 2019-04-11:
TSDuck version 3.17 is out (including this fix).

alevinsn, 2019-04-18:
I tested these changes against the latest code in the git repository as of today (but nothing has changed related to the continuity counter code since 3.17), and from my perspective, this issue is fixed (as of 3.17).  I did notice one thing during an inspection of the code for the ContinuityAnalyzer class, but I will submit that as a separate issue, and I also noticed a misleading message when I ran a TS file with the continuity counter incorrectly incremented when there is no payload through tsfixcc, although the message is reported by the ContinuityAnalyzer class.  I will also submit that as a separate issue.  Note that I tested against both the bad TS file and a good TS file, using tsanalyze and tsfixcc --noaction, and the correct behavior regarding discontinuities is now exhibited by both applications.

closed 2019-04-18 by alevinsn