tsduck/tsduck #257: Feature request:  tsp-like functionality in tsduck library

alevinsn, 2019-04-03:
It would be nice to be able to create an executable, separate from tsp, that can make use of tsduck plug-ins using a plug-in pipeline that is similar to what the tsp executable does.  Some plug-in repository-related classes are already in the tsduck library, but some of them are in the tsp project.  Interestingly, some of these classes are documented in the tsduck developer documentation (like tspInputExecutor, etc), which tends to give the indication that these classes are part of the tsduck library, even though they are not.

Anyway, this isn't a fully formed idea, but I wanted to put the concept out there.  For example, if tsp-like functionality were to be used programmatically, it might be appropriate to set plug-in properties programmatically as well, rather than simulating a command-line for passing in property values.

lelegard, 2019-04-03:
Could be. But there is not one single way to use plugins. The commands `tsp` and `tsswitch` use the same plugins but in a different way. The application logic is different, the executor classes are different.

But you are right on one point: the documentation of the executor classes should not be in the TSDuck library documentation.

alevinsn, 2019-04-03:
I hadn't considered tsswitch and haven't personally used it, but I would suspect that the tsp-style plug-in pipeline in which successive plug-ins are used in sequence is the more common usage scenario and the one that would be of most interest in duplication external to the tsp executable.

But, let's consider a very simple situation:  a developer wants to use the DektecInputPlugin class in their application.  The DektecInputPlugin class is part of the tsduck library--as such, presumably there ought to be a way to use it in one's own software application.  tsplugin_dektec is a very, very thin layer around DektecInputPlugin and DektecOutputPlugin.  But, if a developer wants to use DektecInputPlugin, how should they go about doing so?  How should they go about creating an instance of DektecInputPlugin, since the constructor expects a TSP * input argument, and what to use for an object that implements the TSP class isn't immediately obvious.  The documentation states:  "Each plugin has an associated TSP object to communicate with the Transport Stream Processor main executable."  So, the TSP concept appears to be baked into the plug-in concept and library, even though the actual TSP isn't part of the library itself.

But, even if one figures out the right approach for passing in a TSP object to the constructor, there is no good way to set options on the DektecInputPlugin object programmatically.  The only way the plug-in has of "setting" options is indirectly via the getOptions() method, and the actual settings are declared in the private member _guts, and the Guts class isn't even declared in the .h file.  So, a programmer would be required to somehow simulate a command-line in order to configure the DektecInputPlugin.  And, I should also add that using DektecInputPlugin is the only way to make use of input Dektec devices in tsduck.  So, effectively, we have a class DektecInputPlugin, that is part of the tsduck library, but with no good way to use it as a consumer of the library without going through the tsp executable, which is not part of the library.

Anyway, I think this demonstrates the need for something better, but what "something better" means is still a bit vague.  If I consider something like the various ffmpeg libraries, these can be used independent of the ffmpeg executable, and in some cases, some functionality offered by the libraries isn't easily available via the ffmpeg executable and can generally only be accessed programmatically.

lelegard, 2019-04-04:
Plugin classes indirectly derive from `ts::Args`, the ubiquitous class which handles command line arguments. The actual class (here the plugin class) defines its command line syntax. The command line arguments are provided using the `analyze()` method.

So, to programmatically provide options to a plugin, just use:
~~~
ts::InputPlugin* plugin = new DektecInputPlugin(my_tsp);
...
bool success = plugin->analyze(u"dektec", {u"--device", u"0", u"--frequency", u"123456"});
~~~

The result is false on command line errors. The error messages are displayed through the `TSP` object.

Implementing a `TSP` object is simple. Just define a class which derives from `ts::TSP`. Implement its virtual methods (simple implementations are sufficient). The method `writeLog()` is used to collect (and probably display) log and error messages. The class `ts::PluginThread` is an example of `TSP` subclass which executes a plugin in a thread. This class is the common base for all plugin executors in `tsp` and `tsswitch`. You may use it if it is convenient for you or define another implementation of `TSP`.

Then, you can use your input plugin as follow:
~~~
success = plugin->getOptions() && plugin->start();
...
ts::TSPacket buffer[BUFSIZE];
size_t count = plugin->receive(buffer, BUFSIZE);
if (count == 0) {
    // error or end of stream
}
~~~

alevinsn, 2019-04-04:
OK, that's workable, although the approach of using analyze() to pass in arguments seems a bit klunky to me.  I would understand if this were low priority, but what would be helpful would be to document this in the "Using the TSDuck library" section of the documentation, at https://tsduck.io/doxy/usinglibrary.html .  It would also be helpful to illustrate how to do this in sample-app.  Currently, sample-app has almost no code.  Perhaps sample-app could demonstrate how to use an input plugin, then connect the output of an input plugin to a processing plugin, and finally connect the output of the processing plugin to an output plugin.

lelegard, 2019-04-08:
> the approach of using analyze() to pass in arguments seems a bit klunky to me.

Plugins were designed to be command-line based from the beginning. Having a purely object-based set of getters & setters would require to modify all plugins and maintain two API's for each plugin. I do not feel like going this way. Too much work for a modest result.

> Currently, sample-app has almost no code. 

Definitely. It was only meant to illustrate how to build an application using the TSDuck library (compile, link, etc.) If someone is willing to write some more sophisticated sample application, be my guest.

alevinsn, 2019-04-18:
> Plugins were designed to be command-line based from the beginning. Having a purely object-based set of getters & setters would require to modify all plugins and maintain two API's for each plugin. I do not feel like going this way. Too much work for a modest result.

OK, I understand that, but if there is ever a significant re-design of the API in the future, I hope that my request will be considered.  Plus, I don't think it would require two different APIs.  The FFmpeg model uses a single plug-in API that can be accessed indirectly via the command-line or programmatically.  Here's an example:

`av_opt_set(pVideoCodecContext->priv_data, "profile", "main", 0);`

The nice thing about this sort of approach is it allows individual properties to be set independent of other properties, rather than being required to pass all settings to the plug-in instance in a single call.

> > Currently, sample-app has almost no code.
> 
> Definitely. It was only meant to illustrate how to build an application using the TSDuck library (compile, link, etc.) If someone is willing to write some more sophisticated sample application, be my guest.

I'd be willing to work on this, but if you could provide some pointers for how to hook up a plug-in instance pipeline (input -> processor -> output) programmatically, that would be helpful.

lars18th, 2020-01-24:
Hi @lelegard ,

Are the last commits like https://github.com/tsduck/tsduck/commit/49ea5536093eced6fc3f347ca049e25a6504f3dc (_Refactored tsp and tsswitch in easily accessible C++ classes_) one implementation of this request?

lelegard, 2020-01-24:
> Are the last commits like 49ea553 (Refactored tsp and tsswitch in easily accessible C++ classes) one implementation of this request?

Definitely.

See the source code `tsp.cpp` and the sample application in `sample/sample-app` for details.

lars18th, 2020-01-24:
> > Are the last commits like [49ea553](https://github.com/tsduck/tsduck/commit/49ea5536093eced6fc3f347ca049e25a6504f3dc) (Refactored tsp and tsswitch in easily accessible C++ classes) one implementation of this request?
> 
> Definitely.

Great! :+1: :tada: 

Now I think why not supporting other tstools too. For example: `tsstuff` and `tsresync`. Both generate a Transport Stream and don't have a similar plugin correspondence. So they are good candidates for an API class. Don't you think so?