tsduck/tsduck #259: National font problem in EIT

Taadeus, 2019-04-05:
Hello,
I am starting the adventure of TSDuck and I am determined to at least master operations on EIT tables. I have completed the instructions:
D: \> tsp -I dvb --frequency 562000000 -P eit -P tables -p 18 --pack-all-sections --all-once --xml-output trb.xml -O file trb.ts
and I received an xml file.
Trial:
 D: \> tstabcomp -c --default-charset UTF-8 trb.xml trb.bin
it ended with a series of errors.
[tstabcomp_trb.txt](https://github.com/tsduck/tsduck/files/3047451/tstabcomp_trb.txt)
According to me, this is a problem with Polish fonts in the <text> line. If I exchange these fonts for basic fonts (without Polish "hooks"), errors do not appear. If I shorten the text without changing the fonts, there are also no errors. It only applies to long entries around 230, 240 characters. If I have the number of characters, they are always in the norm <256. Is there a solution to keep the original <text>  record with Polish fonts?
[Inject_trb.txt](https://github.com/tsduck/tsduck/files/3047459/Inject_trb.txt)
[image]
[image]
[XML_trb.txt](https://github.com/tsduck/tsduck/files/3047462/XML_trb.txt)

Thank you for your support now.

Regards Tadeusz

lelegard, 2019-04-05:
> If I shorten the text without changing the fonts, there are also no errors. It only applies to long entries around 230, 240 characters. If I have the number of characters, they are always in the norm <256. Is there a solution to keep the original record with Polish fonts?

The standard says that a descriptor cannot be longer than 255 bytes (the length field is on one byte). In an extended event descriptor, without any item entry, the maximum size of the text field is 249 __bytes__, not characters. If you encode non ASCII unicode code points into UTF-8, they will use more than 1 byte each. So, depending on the number of non-ASCII characters in your text, the maximum number of characters in the text may vary. 230 or 240 are credible values.

lelegard, 2019-04-05:
Could you post a few existing binary sections? Using `-P tables -a -b foo.bin`.

Taadeus, 2019-04-07:
Hi Lelegard,
 Thank you for your answer. I should add that my EIT xml file
(obtained from the national DVBT transponder) starts as follows:
<? xml version = "1.0" encoding = "UTF-8"?>
The result of the compilation gives the same errors whether I apply or not:
--default-charset UTF-8
In the above <text> line there are 220 characters which corresponds to 234 bytes. Therefore, what you wrote above.
  <text> is under 249 bytes.
If I delete 4 basic characters in this line (without Polish hooks) the compilation error disappears.
I attach the file you requested.
[trb.zip](https://github.com/tsduck/tsduck/files/3051336/trb.zip)

Taadeus, 2019-04-07:
starts as follows:
"?xml version="1.0" encoding="UTF-8"?"
...
"text"    is under 249 bytes.

lelegard, 2019-04-07:
OK, the reason is clear now. In the original binary EIT's, the strings are encoded in ISO-8859-2 (Eastern Europe, prefix `10 00 02`). With this encoding, each character uses only one byte, making the storage more compact than UTF-8 (provided that all characters can be encoded, which is the case for the Polish language). 

In some extended event descriptors, the text fits in the descriptor in ISO-8859-2  encoding but not in UTF-8 encoding. The 234 (or so) limit is due to the fact that those descriptors also include "entries", letting less room for the text.

So, you cannot compile these descriptors in UTF-8. Note that XML files are always encoded in UTF-8. I mean that the binary strings in the binary descriptors cannot be in UTF-8. Or you need to split them in several descriptors.

The correct command is, in your case, `tstabcomp --default-charset ISO-8859-2`.

But, wait, not yet 😢 

After that, I discovered an issue with the propagation of the `--default-charset` within the layers of tables and descriptors. I just fixed it. So, you need to pull the repo and rebuild in order to make `tstabcomp --default-charset ISO-8859-2` work correctly.

Taadeus, 2019-04-08:
Thanks a lot, now I'm leaving for a week and I will not be able to work on your answer. However, I thank you very much now. As soon as I come back, I will work on it and give you an answer.

Best Regards Tadeusz
-

lelegard, 2019-04-11:
TSDuck version 3.17 is out (including this fix).

Taadeus, 2019-04-13:
Hi Lelegard,
generally works well but ... :

I'm getting an XML file but with an error message (in tsduck 3.16 there is no error).
2019_04_13_xml_grab.jpg

The received file submits the procedure tstabcomp -c, tstabcomp -d.
I compare the input xml with the output xml.
Results in file:
2019_04_13_tbr.jpg
All rows containing the following are omitted from the output file:
   !-- PID 0x0012 (18) --
Probably are skiping all the spaces at the beginning and end of the lines "text"

Regards Tadeusz

[image]
[image]

Taadeus, 2019-04-13:
One more type of error occurs several times in the file.
2019_04_13_tbr1.jpg
[image]
[trb_in.zip](https://github.com/tsduck/tsduck/files/3076244/trb_in.zip)
[trb_out.zip](https://github.com/tsduck/tsduck/files/3076247/trb_out.zip)

lelegard, 2019-04-13:
> Probably are skiping all the spaces at the beginning and end of the lines "text"

You are right. Leading and trailing spaces were improperly trimmed by default from text nodes while reading XML files. This is fixed now. I just pushed the fix.

Taadeus, 2019-04-16:
Hi Lelegard,
I have learned a little bit of xml write. I delete questions related to the xml record. I would like to receive only your comment about the error:
* Error: dvb: IUnknown ::  ...
Thank you for your help so far.

Regards Tadeusz

lelegard, 2019-04-17:
> Error: dvb: IUnknown :: ...

This is unfortunately a spurious debug message which appears on Windows only. It has been introduced in 3.17 with the support of ATSC tuners on Windows. Basically, without prior knownledge, we try to get the DVB or ATSC tuning request interface of the Windows / DirectShow tuner. One of them fails, which is normal. Reporting the error should be allowed in debug mode only. Not all the time. I just fixed it.

So, you can just safely ignore the message.

Please confirm that the DVB tuner input is correctly working, despite the message.

Taadeus, 2019-04-17:
Hi Lelegard,
Yes, I confirm it. The resulting files are correct. I just moved to Linux, there as you wrote this error does not occur. Subject to be closed. Thank you again for your support.

Best Regards Tadeusz

closed 2019-04-17 by Taadeus