tsduck/tsduck #260: EMMG/PDG <->MUX and datainject plugin

arvindr19, 2019-04-08:


lelegard, 2019-04-08:
> PDG should generate and datainject suppose to update PMT with private data according to CA_system_id and ECM_pid.

Not at all. A CA_descriptor in the PMT is used to signal an ECM PID only. ECM's are inserted by the `scrambler` plugin. So, this plugin adds the CA_descriptor in the PMT. ECM's are related to a service. This is why they are signaled in the PMT of this service.

The EMMG/PDG, on the other hand, generates global data, unrelated to any service. So, there is no point signaling the inserted data in any PMT. If the data are EMM's, then the corresponding PID is signaled in a CA_descriptor in the CAT (which is a global table, not bound to any service). See the [TSDuck user's guide](https://tsduck.io/download/docs/tsduck.pdf) section 5.2.19 for a CAS example. The `cat` plugin is used just before the `datainject` plugin to add the CA_descriptor in the CAT.

If the PDG is used to insert private data, not EMM's, then the PID shall probably be signaled through some private mechanism. But not a CAT.

If the PDG generates private data which are somehow related to one single service, you may want to reference this PID in the PMT of the service. But not in a CA_descriptor. It is reserved for ECM's. In the case of a private data PID in a service, you must add this PID as a component of the service. Important: _never_ reference a PID into a private descriptor, this would not be resistant to PID remapping.

Remember:
- ECM = service-specific data => CA_descriptor in PMT of this service
- EMM = global data => CA_descriptor in TS-wide CAT

That being said, I suppose that this configuration is just for educational purpose. Because using `tsecmg`/`tsmmeg` and also `tsp -P scrambler ... -P datainject ...` makes a completely fake system. In reality, you may want to test a CAS (and you use a real ECMG/ECMG and `tsp` as test tool) or you want to test a MUX (and you use `tsecmg`/`tsmmeg` as test tools and a real MUX).

arvindr19, 2019-04-12:
Hi Thierry,

Thanks for detailed response.

We are developing ECMG and PDG. We have taken TSDuck's ECMG and EMMG as references.

Using our ECMG, we are able to generate an ECM and insert into the TS using the scrambler plugin.

The private data which we are generating is specific to the service and has to be put into PMT as a descriptor. However, when using our PDG, we see that the data is generated properly, but it is not getting inserted into the PMT as a descriptor.

Also, we found that we have to use "-s" to include the data as a section and "-t 1" to consider the data as private data. Accordingly, we have updated our terminal 3 command as follows:

tsemmg -v -s -t 1 --mux localhost:3333 -u localhost:3333 --log-protocol=info --log-data=debug

However, when we specify PMT PID in datainject plugin, it throws the following error:

Error: datainject: data PID conflict, specified 4096 (0x1000), now found as input PID, try another one

If we use a different PID, then the data is getting inserted into a separate table, not into the PMT which we need. Also, tstables shows the data as EMM and not as private data.

How to resolve this conflict? Is there any way to indicate that the private data generated should be inserted as part of an existing PMT?

Regards
Arvind

lelegard, 2019-04-12:
> We are developing ECMG and PDG. We have taken TSDuck's ECMG and EMMG as references.

Take care. `tsecmg` and `tsemmg` are simple test tools. I have already developed a real ECMG years ago and the architecture was quite different. A robust production-grade ECMG shall handle situations which are not implemented in `tsecmg`. Typically, the simple one-client / synchronous dialog which is implemented in `tsecmg` is not sufficient. For instance, the ECMG shall process any unsollicited status request at any time which must be handled differently as expected responses. The real ECMG I developed included multi-client support, full asynchronous design (using a "reactor" design pattern), load management, timeout management, status / keep-alive handling, etc.

> The private data which we are generating is specific to the service and has to be put into PMT as a descriptor. However, when using our PDG, we see that the data is generated properly, but it is not getting inserted into the PMT as a descriptor.

I suspect a misunderstanding here. What does your PDG generate?

- A stream of sections to be inserted in a dedicated PID (ie. the PID contains exactly what you are inserting, nothing less, nothing more). And you want to reference this PID (its PID _number_) in the PMT.
- Some data which _are_ descriptors to be inserted in the PMT.

The EMMG/PDG <=> MUX interface is designed for the first case. The second case is meaningless.

> However, when we specify PMT PID in datainject plugin, it throws the following error:

Because you ask the plugin to _create_ the entire content of the PID with the injected data. And since this is the PMT PID, the PID already exists.

arvindr19, 2019-04-15:
Hi Thierry,

Thanks for prompt response.

> I suspect a misunderstanding here. What does your PDG generate?
> * Some data which _are_ descriptors to be inserted in the PMT.
Our PDG generates private data which is a part of CA_descriptor in PMT.
CA_descriptor has format tag/tag_length/CA system id/CA pid/**private_data**
We need to put private data into PMT's CA_decriptor.
 Is there any way to indicate that the private data generated should be inserted/updated as part of an existing PMT's CA_decriptor?
> The EMMG/PDG <=> MUX interface is designed for the first case. The second case is meaningless.
Is there any way around or TSDuck plugin to generate private data and insert it into PMT's CA_decriptor?

Regards
Arvind

lelegard, 2019-04-15:
OK, so I see where the misunderstanding is. A PDG (in the sense of DVB SimulCrypt) is designed to generate a stream of data, meaning the complete content of a dedicated PID, typically sections but in fact whatever you want if you work in packet mode. A PDG _cannot_ generate private data for a CA descriptor in a PMT. This is not a limitation of TSDuck; this is simply incompatible with the DVB SimulCrypt head-end architecture (see ETSI TS 103 197).

To insert private data in the CA descriptor for your ECM PID, use option `--private-data` in the `scrambler` plugin. 

But note that these private data must be mostly static. They cannot change too often. Each time you change them, you must generate a new PMT, cycling the version number, etc. Typically, changing the CA_descriptor private data either occurs ahead of access criteria transition or simply never occurs. See annex G in ETSI TS 103 197. If you want to change the private data of the CA descriptor too often, then you probably have a system design issue.

arvindr19, 2019-04-15:
Hi Thierry,
Thank you so much for the clarification.
In ETSI TS 103 197 V1.5.1 section 4.3.3 PDG ⇔ MUX, it is mentioned as below:
> The interface is identical to the EMMG ⇔ MUX interface and may be used for providing CA related private data to the MUX for insertion in the transport stream. This interface is optional; if implemented it shall be implemented as per the EMMG ⇔ MUX interface

Our understanding is CA related private data is of PMT's CA descriptor. Are we missing anything?

If we use "-s" and "-t 1" as options with the tsemmg plugin, it will generate sections.
Can't we insert these sections in existing PMT instead of specifying new PID?

Regards
Arvind

lelegard, 2019-04-15:
> Our understanding is CA related private data is of PMT's CA descriptor. Are we missing anything?

In general, a CA system may generate a lot of different forms of private data. The private data in the CA descriptor is just one form of private data. But this kind of data _cannot_ be inserted using the EMMG/PDG <=> MUX interface. The PDG, in the sense of DVB SimulCrypt, can only insert a complete PID of private sections, typically EMM's which are different forms of CA private data.

In fact, this interface was primarily designed to inject EMM's. But, unlike the ECMG <=> SCS interface which is very specific to ECM insertion, the EMMG <=> MUX interface happened to have nothing specific to EMM's. So, it was somewhat useful to declare it more generic as Private Data Generator in case someone would like to insert application-specific data such as HbbTV or DSM-CC carousel. 

> If we use "-s" and "-t 1" as options with the tsemmg plugin, it will generate sections. Can't we insert these sections in existing PMT instead of specifying new PID?

Don't try to use `tsemmg` to do what you want. You went too far the wrong way in your analysis. You need to step back and reconsider your design.

First question: Are your private data really associated to an ECM stream?

- If yes, go on.
- If no, your system design is wrong and reconsider it.

Second question is: Are your private data static? Meaning do you want to insert the same binary blob all the time in the CA_descriptor for a given ECM stream?

- If yes, use option `--private-data` in `scrambler` plugin.
- If no, your system design is wrong and reconsider it.

closed 2019-06-04 by lelegard