tsduck/tsduck #261: Very minor request:  include dates in CHANGELOG.txt

alevinsn, 2019-04-13:
Instead of just putting "VERSION 3.17-1189", the request is to do something like this:  "VERSION 3.17-1189 (<date of release>)".  This would make it slightly easier to review the change log for new changes since the last time I've reviewed the file.

lelegard, 2019-04-14:
I am afraid I do not understand the request. What difference does "()" make?

I (try to) update the commit number after the dash each time the file CHANGELOG.txt is updated. So, each time the file changes, the version is updated. Now, to find fine-grained differences from any other version, we still have `git` commands and associated GUI's.

alevinsn, 2019-04-17:
I know that this can be accomplished using git commands.  I was looking for something a little quicker and a little easier.  I'm not suggesting that someone go through all existing changelog messages and modify them, but to do so when a new version is released.  The CHANGELOG.txt file is packaged with each release of the software, and I suspect that there are plenty of users that get the file by downloading the ZIP file for a particular release, not by checking it out using git.  So, this would provide them with an option as well for a date stamp for each version.  My suggestion of wrapping parentheses around the date was just to make it look a little better in my opinion, but it really doesn't matter as long as the date shows up in some form.

closed 2019-05-26 by lelegard