tsduck/tsduck #264: multicast and

alaaodeh, 2019-04-17:
Hi,

I have 3 streams scrambled channels 

    stream1 => 239.255.1.1 port 59000
    stream2 => 239.255.1.2 port 59000
    stream3 => 239.255.1.3 port 59000

and i need to modulator to output dektec DVB-S2 107 rev 4 
and I need the full command please using tsduck

rezna11, 2019-04-18:
The closest you can get is probably using `null` input with sufficient bitrate to cover all 3 multicasts and to match DVB-S/S2 output parameters, followed by `regulate` plugin and then use `merge` plugin 3 times (one for each multicast input). 

Honestly I'm not sure if TSDuck added some magic options lately to improve the output stability, if I remember right, there were some issues with PCR/DTS/PTS on the output, since TSDuck was not meant to work like a mux. Every time I used merge, it worked good first but over time (couple of hours), the output would degrade, not being decoded properly.

alaaodeh, 2019-04-19:
Thanks for your kindly support,

could you send me sample command please to test it ?

closed 2026-04-23 by github-actions[bot]