tsduck/tsduck #265: ContinuityAnalyzer class in generator mode may use an algorithm that is too simple

alevinsn, 2019-04-18:
When the ContinuityAnalyzer class is configured to run in generator mode, it uses a simple approach:  it clears all discontinuities and always increases the CC count unless the current packet doesn't have a payload.  This will likely work fine the majority of the time.  But, let's consider a potential case with the ContinuityAnalyzer used in generator mode with the rmsplice plug-in.  Let's say that there is one splice event sequence in an input stream, corresponding to a commercial break.  So, if the fixcc option is enabled, what one would expect the rmsplice plug-in to do in this case is to output all the packets, exactly as they appear in the input stream, until it hits the splice out event.  Then it would drop all packets until getting a splice in command, at which point it would start outputting packets again, but with the CC adjusted appropriately and also time stamps adjusted appropriately if the plug-in has been configured to do so.

However, the expected behavior is not guaranteed to occur due to the way the ContinuityAnalyzer works in generator mode.  Just considering all the packets prior to the splice out command, if any of these packets have the discontinuity indicator turned on, this will be lost.  Further, if there are any duplicate packets, that is, successive packets with the same content and with the CC the same value, this will also be lost, and that's perhaps a more significant issue.

After the splice, there is no choice but to adjust the CC (if fixcc is enabled), but this can be done in a fashion that respects both the discontinuity indicator and duplicate packets.  Let's say that, in generator mode, the ContinuityAnalyzer class did not reset the discontinuity indicator.  In that case, after the splice, the first packet for each monitored PID could have its discontinuity indicator turned on by the rmsplice plug-in.  Then, when the ContinuityAnalyzer sees any packets with the indicator turned on, it would act as a reset, resetting the current fixcc value to that of the packet with the indicator turned on.

To handle duplicate packets, it would be necessary to keep track of the last packet for each PID.  Then, when in generator mode, if it sees that the CC value is the same for the current packet as that of the last packet (the original version of the packet, not the version with its CC value massaged), it would compare the memory of each packet, and if they are the same, it would maintain this by setting the CC value of the current packet to be the same as used for the last packet.  Further, any time a discontinuity packet is observed would reset the duplicate packet checking state.

Anyway, I think the approach that I suggested would properly address the possibility of expected discontinuities and duplicates when the ContinuityAnalyzer class is in generator mode.

lelegard, 2019-04-20:
Thanks for reporting this. There are a few more corner cases I am thinking about. I will fix this.

lelegard, 2019-09-11:
Fixed in commit b53e6c751e237040eb54d013fc5ed297fa1dd98a