tsduck/tsduck #266: ContinuityAnalyzer message for packets without payload but with the CC incremented is confusing

alevinsn, 2019-04-18:
I deliberately generated a TS with the CC incremented for packets without any payload and ran it through tsfixcc --verbose --noaction <filename>.  tsfixcc correctly reported discontinuities, but it did so in a confusing fashion.  Here is an example of the output:

```
packet: 1,347,899, PID: 0x0164, missing:  0 packets
<snip>
2,493,804 packets read, 36 discontinuities, 0 packets updated
```

The "packet:" message is reported by the ContinuityAnalyzer class (in the feedPacketInternal() method), and even though I know that this message corresponds to an unexpected discontinuity caused by the CC value being incremented for a packet without any payload, I have no idea what this message means.  This message appears to be used for multiple possible situations, not just for the no payload situation, so perhaps it would be useful to have different messages for each of the different issues that might occur.

lelegard, 2019-04-20:
Yes, this is confusing indeed. In case of a detected discontinuity, the code computes the number of missing packets (modulo 16). For instance, if we get CC values 7 and 10, we know that 3 packets are missing. In the case of 7 and 8 without payload, the same algorithm reports 0 missing packets while the message should be different. I will fix it.

lelegard, 2019-04-20:
Fixed in commit cbd8859e71c93e809b2a8e1b52e07f8be1f8f78e

closed 2019-04-20 by lelegard

alevinsn, 2019-04-23:
I know that you verified that this bug is fixed, but I can also confirm that it is verified.  I'd say now TSDuck offers the best CC checking of any TS tool on the market (free or non-free) of which I have experience with.  Two different pieces of software that I've used have reported that there is a CC error in this case, but neither reported why.  The new message is very clear, and if someone doesn't understand exactly what it means, it would be fairly quick to do a search on the Internet and find out.  BTW, nice job on abstracting out the code having to do with CC and creating the ContinuityAnalyzer class.