tsduck/tsduck #269: Compile errors (Debian 7 and Debian WSL)

lars18th, 2019-04-19:
Hi @lelegard ,

Last version doesn't compile, almost with Debian 7 (I know it's deprecated) and Debian WSL (in Windows 10).

- Debian 7:
```
In file included from tsAbstractTable.h:287:0,
                 from tsAbstractLongTable.h:36,
                 from tsAIT.h:36,
                 from tsAIT.cpp:34:
tsAbstractTableTemplate.h: In instantiation of 'ENTRY& ts::AbstractTable::EntryWithDescriptorsMap<KEY, ENTRY, <anonymous> >::operator[](const KEY&) [with KEY = ts::ApplicationIdentifier; ENTRY = ts::AIT::Application; typename std::enable_if<std::is_base_of<ts::AbstractTable::EntryBase, ENTRY>::value, void>::type* <anonymous> = 0u]':
tsAIT.cpp:133:49:   required from here
tsAbstractTableTemplate.h:99:118: error: 'class ts::AbstractTable::EntryWithDescriptorsMap<ts::ApplicationIdentifier, ts::AIT::Application>' has no member named 'emplace'
tsAbstractTableTemplate.h: In member function 'ENTRY& ts::AbstractTable::EntryWithDescriptorsMap<KEY, ENTRY, <anonymous> >::operator[](const KEY&) [with KEY = ts::ApplicationIdentifier; ENTRY = ts::AIT::Application; typename std::enable_if<std::is_base_of<ts::AbstractTable::EntryBase, ENTRY>::value, void>::type* <anonymous> = 0u]':
tsAbstractTableTemplate.h:100:1: error: control reaches end of non-void function [-Werror=return-type]
cc1plus: all warnings being treated as errors
make[2]: *** [release-x86_64/tsAIT.o] Error 1
```

```
# gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/4.7/lto-wrapper
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Debian 4.7.2-5' --with-bugurl=file:///usr/share/doc/gcc-4.7/README.Bugs --enable-languages=c,c++,go,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-4.7 --enable-shared --enable-linker-build-id --with-system-zlib --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --with-gxx-include-dir=/usr/include/c++/4.7 --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --enable-gnu-unique-object --enable-plugin --enable-objc-gc --with-arch-32=i586 --with-tune=generic --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc version 4.7.2 (Debian 4.7.2-5)
```

- Debian WSL:
```
In file included from /usr/include/c++/6/bits/move.h:57:0,
                 from /usr/include/c++/6/bits/stl_pair.h:59,
                 from /usr/include/c++/6/bits/stl_algobase.h:64,
                 from /usr/include/c++/6/bits/char_traits.h:39,
                 from /usr/include/c++/6/string:40,
                 from tsPlatform.h:655,
                 from tsMPEG.h:37,
                 from tsAbstractDefinedByStandards.h:36,
                 from tsAbstractSignalization.h:36,
                 from tsAbstractDescriptor.h:36,
                 from tsAACDescriptor.h:36,
                 from tsAACDescriptor.cpp:34:
/usr/include/c++/6/type_traits: In instantiation of 'struct std::underlying_type<unsigned char>':
tsxmlElement.h:426:107:   required by substitution of 'template<class ENUM, typename std::enable_if<std::is_enum<_Tp>::value>::type* <anon
ymous>, class INT> bool ts::xml::Element::getIntAttribute(ENUM&, const ts::UString&, bool, ENUM, INT, INT) const [with ENUM = unsigned cha
r; typename std::enable_if<std::is_enum<_Tp>::value>::type* <anonymous> = <missing>; INT = <missing>]'
tsAACDescriptor.cpp:187:88:   required from here
/usr/include/c++/6/type_traits:2256:38: error: 'unsigned char' is not an enumeration type
       typedef __underlying_type(_Tp) type;
                                      ^~~~
../../Makefile.common:898: recipe for target 'release-x86_64/tsAACDescriptor.o' failed
make[2]: *** [release-x86_64/tsAACDescriptor.o] Error 1
```

```
$ gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/6/lto-wrapper
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Debian 6.3.0-18+deb9u1' --with-bugurl=file:///usr/share/doc/gcc-6/README.Bugs --en
able-languages=c,ada,c++,java,go,d,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-6 --program-prefix=x86_64-linux-gnu- --enable-shar
ed --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --with-
sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-o
bject --disable-vtable-verify --enable-libmpx --enable-plugin --enable-default-pie --with-system-zlib --disable-browser-plugin --enable-ja
va-awt=gtk --enable-gtk-cairo --with-java-home=/usr/lib/jvm/java-1.5.0-gcj-6-amd64/jre --enable-java-home --with-jvm-root-dir=/usr/lib/jvm
/java-1.5.0-gcj-6-amd64 --with-jvm-jar-dir=/usr/lib/jvm-exports/java-1.5.0-gcj-6-amd64 --with-arch-directory=amd64 --with-ecj-jar=/usr/sha
re/java/eclipse-ecj.jar --with-target-system-zlib --enable-objc-gc=auto --enable-multiarch --with-arch-32=i686 --with-abi=m64 --with-multi
lib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --t
arget=x86_64-linux-gnu
Thread model: posix
gcc version 6.3.0 20170516 (Debian 6.3.0-18+deb9u1)
```

I report here only FYI.
Regards.

lelegard, 2019-04-19:
Both versions of GCC are either too old (4.7.2) or broken (6.3.0).

GCC 4.7.2 is too old. The oldest GCC I tested is 4.8.5, as found on RHEL / CentOS 7.6.

GCC 6.x and 7.x are broken. See #231 and #219.

lars18th, 2019-04-19:
OK.
Thank you!

closed 2019-04-19 by lars18th