tsduck/tsduck #270: Plugin timeref: Add "eit" processing

lars18th, 2019-04-21:
Hi,

After reviewing the documentation I see that the request #258 doesn't have sense.
This is because the `timeref` plugin does _all_ time manipulations. However, at time it doesn't have support to manipulate (aka update) the EIT tables.

So, I request to update this plugin with this new parameter:
- `-P timeref --eit`: Update EIT tables with the new timeref (by default off).

The behaviour is this: it calculates the difference between the original time reference and the new one, and it adds the calculated offset.

You agree?

lelegard, 2019-05-04:
Implemented in commit 5a63f1d9bd78ce1aac1b7a6e61e6f645f6ac3196

lars18th, 2019-05-10:
Hi,

Tested and working.
Thank you @lelegard ! :wink:

closed 2019-05-10 by lars18th