tsduck/tsduck #274: Request: Plugin PMT add user descriptors

lars18th, 2019-04-24:
Hi @lelegard ,

Regarding this problem that I found in my last PR (don't worry, the error described doesn't exist):
https://github.com/tsduck/tsduck/pull/271#issuecomment-486209720

However, I discovered that the command `-P pmt -a 7777/0x06 --add-programinfo-id 0x4B4C5641` described in the example doesn't work (for the specified objective). With this command a new decriptor "Registration" is added **to the PMT table**, and not to the PID used. So, at time it's impossible to set correctly the type of the stream using this plugin.

I suggest then to add a new parameter to the plugin in this way:
- `-P pmt --set-generic-descriptor pid/id/data`: where < pid > is the pid where add the descriptor, < id > is the descriptor id (i.e. 0x05=Registration; 0x52=Stream Identifier; 0x26=Metadata") and < data > is the binary representation of the Data (i.e. 0x4B4C5641 that's KLVA).

With this functionality then we can use `-P pmt -a 7777/0x06 --add-generic-descriptor 777/0x05/0x4B4C5641` and it will really work.

Please, can you implement it?
Thank you!

lars18th, 2019-04-26:
Hi @lelegard ,

Please, can you just answer this question:
- It's possible at time with the current TSDuck set the correct value for the KLVA stream?

If yes, then explain how to do it.
If not, then we need to wait.

Just to find out if there's any other option.
Thank you your time!

lelegard, 2019-05-03:
Hi,

This is done in commit 8c53378e62d1c4f412d28c9e912dfc62491c6a96.

The added and modified options are:
~~~
--add-registration id
    Add a registration_descriptor in the program-level descriptor list in the PMT.
    The value is the format_identifier in registration_descriptor, e.g. 0x43554549 for CUEI.

--add-pid-registration pid/id
    Add a registration_descriptor in the descriptor list of the specified PID in the PMT.
    The value is the format_identifier in registration_descriptor, e.g. 0x43554549 for CUEI.

--add-programinfo-id id
    A legacy synonym for --add-registration.
~~~

lars18th, 2019-05-08:
Hi @lelegard ,

It works very well. Thank you!

However, before close this issue you need to update the documentation for the PES encapsulation. The current example is wrong, and the values are different based on SYNC or ASYNC. Here the correct values:

- ASYNC example (default):
```
tsp ... \
 -P encap -o 7777 --pes-mode ... \
 -P pmt -s 100 -a 7777/0x06 --add-pid-registration 7777/0x4B4C5641 \
 ...
```

- SYNC example (with pcr):
```
tsp ... \
 -P encap -o 7777 --pes-mode --pes-offset=-50000 --pcr-pid 101 ... \
 -P pmt -s 100 -a 7777/0x15 --add-pid-registration 7777/0x4B4C5641 \
 ...
```

So type is "0x06" for Asynchronous and "0x15" for Synchronous.

Regards.

lelegard, 2019-05-08:
OK, doc updated. You clearly understand this encapsulation format better than I do.

Please close this issue if you consider it is complete.

lars18th, 2019-05-10:
> OK, doc updated.

Great!
However, you forget some part of the documentation. See in page 134 the section before the example. You need to upgrade it to include the Synchronous mode.
Please, copy the full description (updated un my last PR) from the source code:
https://github.com/tsduck/tsduck/blob/e640cffbcd1cfa54de51d3fc12ddd4a708d2506e/src/libtsduck/tsPacketEncapsulation.h#L68

In particular the missing part is these two paragraphs (even some other parts are updated):
https://github.com/tsduck/tsduck/blob/e640cffbcd1cfa54de51d3fc12ddd4a708d2506e/src/libtsduck/tsPacketEncapsulation.h#L111

> Please close this issue if you consider it is complete.

When completed! :wink:

lelegard, 2019-05-10:
Documentation updated in commit 3b27b2ba8f23d9e9fbb3c68a5a8f1014ec27778f.

lars18th, 2019-05-11:
> Documentation updated in commit [3b27b2b](https://github.com/tsduck/tsduck/commit/3b27b2ba8f23d9e9fbb3c68a5a8f1014ec27778f).

Thank you! I close the issue.

closed 2019-05-11 by lars18th