tsduck/tsduck #276: * Error: datainject: error receiving from UDP socket: Bad file descriptor

adambever1, 2019-04-25:
Hi Thierry,

I'm trying to replicate the example in your documentation of the complete conditional access system test bed (5.2.19) using a static transport stream. Attached you'll find the script I'm using and the output of the tsemmg and the output of the script. 

[script.txt](https://github.com/tsduck/tsduck/files/3114700/script.txt)
[output of tsemmg.txt](https://github.com/tsduck/tsduck/files/3114701/output.of.tsemmg.txt)
[output script.txt](https://github.com/tsduck/tsduck/files/3114702/output.script.txt)

I'm successfully able to scramble the stream using tsecmg however I'm not seeing the EMM stream (I verified using tsanalyze). The final error I get is 
` Error: datainject: error receiving from UDP socket: Bad file descriptor`

Any help would be greatly appreciated.

Best Regards,

Adam

lelegard, 2019-05-02:
Hi,

First, the missing EMM stream: the most usual cause for not seeing anything that should be inserted is the lack of stuffing in the input stream. This is so common that I described it in the TSDuck documentation, with possible solutions. See the reference page for `tsp`, especially the section named "Modifying, inserting and deleting packets".

Do you see the ECM stream? If there is no input stuffing, you should not see them either.

Second, the error message is a spurious one that appears when the `datainject` plugin terminates and `--udp` was specified. I just pushed the fix. In the meantime, you may safely ignore this message.

In details, the UDP reception of EMM's is performed in a thread which loops on incoming UDP messages. The main plugin thread, when a termination is requested, closes the UDP socket to unblock the receiving thread which then terminates. Of course, the receive operation terminates on error because the socket was closed. But this is an "expected error" which should not be displayed.

adambever1, 2019-06-18:
Hi Thierry,

Using --add-input-stuffing with the tsp plugin did the trick. There was a great example on page 245 of your documentation, under the section 5.2.24.2 Cue insertion in offline files.

Thanks a bunch for your help! 

Best Regards,

Adam

closed 2019-06-18 by adambever1