tsduck/tsduck #277: DVB-S2 multistream support

rezna11, 2019-04-28:
Hi, I'm wondering how to tune in the multistream transponders using TSDuck. It seems it's not possible at all, right? Because it's not supported by Windows and Linux DVB API? Or am I missing something?

lelegard, 2019-05-02:
Hi,

TSDuck cannot do more than what the underlying O/S can provide.

I think there is something on Linux. This is the same parameter as the PLP id with DVB-T2. This is also dependent on the capabilities of the hardware tuner. This should be indicated as "multistream" in the output of `tslsdvb -v`.

On the other hand, I do not see anything on Windows. The PLP selection is not available on Windows either BTW; this is indicated in the TSDuck documentation. 

I could add the DVB-S2 stream-id selection on TSDuck, available on Linux only, just like the DVB-T2 PLP. The most difficult part will be to find a transponder with multiple streams and someone to test it with the appropriate tuner on a Linux box.

rezna11, 2019-05-02:
Hello Thierry. There are multistream transponders on 5W, carrying all French TNT and Italian DTT channels. In theory TSDuck could support multistream on Windows if it could use StreamreaderEx-BDA device as input but I understand that's probably not something you'd be interested in if Linux can handle multistreams natively.
Stream ID selection is one thing but I think you would also have to implement PLS.
I have 2 TBS tuners that can handle multistream, I can receive 5W and running Linux to test this is not a problem, so should you decide to implement this, I can test this.

lelegard, 2019-05-02:
OK, I will "blindly" implement it and let you know when it is ready. Then, please test it and report feedback.

lelegard, 2019-05-10:
Hi @rezna11,

I have added the multi stream selection on DVB-S2 for Linux only. Currently, I did not change the command line options and documentation, only internal code. So, it comes as a "hidden" option using `--plp` which is normally used for DVB-T2 PLP (same multistream concept).

Please pull, rebuild and test if `--plp` can trigger the right DVB-S2 stream. If if works, then I will update the command line syntax with a better `--stream-id` option or equivalent, also replacing `--plp` with DVB-T2. And I will update the documentation.

Also check the description of the tuner with the command `tslsdvb -v`. If the Linux driver can handle multistream on this tuner, you should see "multistream" in the list of tuner capabilities.

rezna11, 2019-05-10:
It's working with TBS 5925. But only one transponder at 5W (12543H), the others don't work because they're also using PLS, which is very common with multistreams. So to really support tuning in multistreams, it's really necessary to also add PLS (Root, Gold) code support.

lelegard, 2019-05-10:
Hi @rezna11,

Thanks for the feedback, I will make it more permanent and documented.

Concerning PLS, I do not know how to enable it. Googling a bit, I see references to "zap-s2 with PLS support", CrazyCat dvb drivers with PLS support, etc. But when I look into the corresponding source code, I see nothing relevant.

Do you know exactly what is the current status of PLS support on Linux?

rezna11, 2019-05-10:
I haven't looked into it too deeply and honestly I'm confused. I know that I tune in MIS with PLS using dvblast, using a command like this:

`dvblast -u -d 232.1.1.1:7000 -i 1 -a 1 -S 3 -f 12585000 -s 35000000 -v 18 -m psk_8 --multistream-id-pls-mode ROOT --multistream-id-pls-code 8 --multistream-id-is-id 1`

so that's stream 1, using ROOT 8 for PLS.

The problem is I don't know why it works. I don't remember what I did when I first installed this Linux machine. It's running CentOS and I did compile TBS drivers but I don't remember how :( It was one of these... https://github.com/ljalves/media_build or https://github.com/ljalves/linux_media or https://bitbucket.org/CrazyCat/linux_media/src/latest/
But I don't know if that's just drivers/firmware for the tuners to support multistream or if it actually modified the Linux DVB API to add multistream with PLS.

lelegard, 2019-05-10:
I have seen these options too from dvblast. But it is confusing:

- I install dvblast on Fedora 30 from the official repo, I get the PLS options.
- I clone the "official" origin git repo from git://git.videolan.org/dvblast.git and there is no PLS option.
- I download the dvblast source RPM from Fedora 30 repo and I get the PLS options.

I will have a look into it. If there is a "standard" way to interface Linux DVB API, then I will use it. If there is a hack into the drivers and the Linux DVB API (even changing some API values as I saw in the CrazyCat drivers), then I will not do it because this will break the tuning for all other TSDuck users.

lelegard, 2019-05-10:
OK, I got it. In fact, there is nothing at driver level. In the source code of dvblast (Fedora version), the PLS code, PLS mode and IS id are just values which are used to compute the stream id.

Either you provide a stream id (option `--multistream-id`) or you provide a triplet of `--multistream-id-pls-mode` (ROOT=0, GOLD=1, COMBO=2), `--multistream-id-pls-code` (0-262143), `--multistream-id-is-id` (0-255).

In the triplet case, the stream id is computer as follow: `( pls_mode << 26 | pls_code << 8 | is_id )`.

In both cases (single value or triplet), the final value is passed as stream id (currently option `--plp` in TSDuck). So, you could probably try now using a manually computed stream id.

I am not qualified to say it this is correct or not but this is what dvblast is doing.

I will implement this and you will have to try.

rezna11, 2019-05-11:
That makes sense, dbvlast debug prints this:

`debug: tuning DVB-S frontend to f=12585000 srate=35000000 inversion=-1 fec=999 rolloff=35 modulation=psk_8 pilot=-1 mis=2049 /pls-mode: ROOT (0) pls-code: 8 is-id: 1 /`

so the MIS (PLP) value is 2049, that's also what Astra Cesbo computes it to, so it seems to be the way how this is handled in Linux

rezna11, 2019-05-11:
This should work, I just changed the soruce to allow PLP greater than 255 and used 98139138 to match stream 2 in PLS Gold 121212 and then TSDuck tuned in one of the TNT streams :)

[image]

lelegard, 2019-05-11:
Thanks for the confirmation. Proper command line options have been added and documented (see below). Please test and close this issue if you consider it is complete.

`--isi value`
Used for DVB-S2 tuners only.
Specify the Input Stream Id (ISI) number to select, from 0 to 255. Used with multistream, see also options --pls-code and --pls-mode.
The default is to keep the entire stream, without multistream selection.
Warning: this option is supported on Linux only. Currently, Windows provides no support for multistream.

`--pls-code value`
Used for DVB-S2 tuners only.
Specifiy the Physical Layer Scrambling (PLS) code value, from 0 to 262143 (0x3FFFF). Used with multistream, see also option --isi.
Warning: this option is supported on Linux only. Currently, Windows provides no support for multistream.

`--pls-mode mode`
Used for DVB-S2 tuners only.
Specify the Physical Layer Scrambling (PLS) mode. Used with multistream, see also option --isi. Must be one of "COMBO", "GOLD", "ROOT". The default is ROOT.
Warning: this option is supported on Linux only. Currently, Windows provides no support for multistream.

rezna11, 2019-05-11:
Thank you, it's working.

closed 2019-05-16 by lelegard

soufiane2020, 2020-05-18:
hello can someone help me to work multistream in astra cesbo