tsduck/tsduck #278: Help: How to identify a packet and execute an action

lars18th, 2019-04-29:
Hi @lelegard ,

Please, can you help me by explaining how to achieve this goal?

- Inside a PID a need to identify duplicated packets. A duplicated packet is any packet that the next packet in the pid is identical (all bytes are equal), so no differences. And the goal is to change the pid number of the _duplicated_ packet. So, the first continues without changes and the next will change to a new pid.

Any idea how to get it?
Thank you for your time.

lars18th, 2019-04-29:
Hi @lelegard ,

Another option more generic can be this:
- Provide a "check" plugin that finds if a packet has some values (for example `0x81;0x11;0x00` at given offset) and then it executes an action (in this case change pid). One example:
`-P check-packet -pid 100 --label 0x81;0x11;0x00 --label-offset 35 --change-pid 101`

And the options can be `--change-pid`, `--remove`, `--duplicate`, `--patch` (where "patch" is like "label" and "label-offset" to change content of the packet).

What you think?

alaaodeh, 2019-05-04:
Hi @lelegard ,@lars18th

Kindly I used the command 

 -P check-packet -pid 100 --label 0x81;0x11;0x00 --label-offset 35 --change-pid 101
but required file and i need the input from DVB s2 signal and drop specific packet

lelegard, 2019-05-16:
Options `--pattern`, `--search-payload` and `--search-offset` have been added to plugin `filter`. See #296.

lars18th, 2019-05-21:
OK. I close this issue, and I continue inside #296 .

Thank you!

closed 2019-05-21 by lars18th