tsduck/tsduck #279: compilation error

lars18th, 2019-04-29:
Hi,

I know, I know it! The list of compilers accepted for the last releases is reduced.
But, I'm trying to complie in unsupported platforms (WSL and old Debians) using "backported GCC 4.9" and "custom compiled GCC 8.2".

More or less, I'm advancing but now I have a problem with the AR process. It seems that it enters inside a loop. This is the error:
```
root@pc:~/tsduck/development/tsduck# make NOTEST=1 NOPCSC=1 NODTAPI=1
  [AR] release-x86_64/libtsduck.a
gcc-/gcc-4.9/bin/ar: error trying to exec '/gcc-4.9/bin/ar': execvp: Cannot allocate memory
```
And if I check the processes with top I see that:
- The memory and virtual memory are exhausted.
- A lot of multiple process are created with _REPEATED_ parameters. This is part of a line of just _one_ process:
`/gcc-4.9/bin/ar --plugin /usr/lib/gcc-4.9-backport/libexec/gcc/x86_64-linux-gnu/4.9/liblto_plugin.so --plugin /usr/lib/gcc-4.9-backport/libexec/gcc/x86_64-linux-gnu/4.9/liblto_plugin.so --plugin /usr/lib/gcc-4.9-backport/libexec/gcc/x86_64-linux-gnu/4.9/liblto_plu...`

So, I'm sure it's a bug, as the call parameters enter in a recuersive call until the process collapses.
Any idea to fix it?

lars18th, 2019-04-29:
Last lines when compiling with `make --debug=all`:
```
    Finished prerequisites of target file `release-x86_64/libtsduck.a'.
   Must remake target `release-x86_64/libtsduck.a'.
Putting child 0x00ec99b0 (release-x86_64/libtsduck.a) PID 5532 on the chain.
Live child 0x00ec99b0 (release-x86_64/libtsduck.a) PID 5532
  [AR] release-x86_64/libtsduck.a
```

And from here the recursive bug.

lars18th, 2019-04-29:
Hi,

Identical problem in a old Debian and in Debian-WSL. In both systems (very different, in one is GCC 4.9 backported and in other is GCC 8.2 compiled in host), the same problem.

Any idea?

lars18th, 2019-04-30:
Hi,

I resolved this problem, as the cause is an incorrect configuration of the compiler environment. So not needed to change/replace/patch the TSDuck source. Great!

However, I suggest to @lelegard to update the building instructions at:
https://tsduck.io/doxy/building.html

Please, comment that to compile using _another_ compiler you need to use, for example:
`make GCC=gcc-4.9 CXX=g++-4.9 AR=gcc-ar-4.9`

The required vars are only `GCC`, `CXX` and `AR`. You doesn't need more; and you can point binaries in the path or with the full path (like `GCC=/usr/local/gcc-8.2.0/bin/gcc`);

lars18th, 2019-04-30:
Hi @lelegard ,

I suggest to comment too about how to use a supported compiler in Debian releases without GCC 8 or 9. That's has sense as a **lot** of versions with more than two years use GCC 6 or 7, and not the GCC 4.2. This includes WSL Debian (Windows 10), Debian 9 Stretch, Debian 8 Jessie and Debian 7 Whezzy.

Here one solution:
Install the *.deb packages of this repository manually:
http://mxrepo.com/mx/repo/pool/main/g/gcc-4.9/
You'll get the `gcc-4.9` in your path without disturbing your native default compiler.

This is the more simple solution to compile last versions of TSDuck.
And it's my solution to compile it in Windows to run inside WSL.

Regards.

lelegard, 2019-05-02:
You are right, this is the simplest method. I tested it on Rasbian 9. The version 4.9 of GCC is in the official repo, no need to get third-party packages, simply `apt install gcc-4.9 g++-4.9`

I do not know if this is also the case for Debian 9. And I have no idea about WSL, I have never used this hybrid monster.

I have updated the Doxygen file with the build instructions accordingly, including your make command: `make GCC=gcc-4.9 CXX=g++-4.9 AR=gcc-ar-4.9 NOTEST=true `

lars18th, 2019-05-03:
Hi @lelegard ,

Thank you for the upgraded documentation.

Let me to add one some comments:
- I use to compile `make GCC=gcc-4.9 CXX=g++-4.9 AR=gcc-ar-4.9 NOTEST=1 NOPCSC=1 NODTAPI=1` and I suggest to add too the `NOPCSC=1 NODTAPI=1` in the documenation. Why? Because it simplifies the compilation/dependencies and a lot of users don't use PSCS and Dektek API.
- I confirm it compiles too in regular Debian 9 x64 with GCC 4.9. However the package is not every time accesible with the `apt-get` command. It depends on the listed repositories.
- In WSL it compiles too. It's quite simple to setup (Instal Debian from the Windows Store). Then your WSL Debian evironment it's a regular Debian 9 (in fact, the packages are the same, it's binary compatible). So, you only need to install the requeriments to compile TSDuck. The only limition with these binaries is that you need to execute the tsp command from the WSL shell (and I prefer it to the CMD/PowerShell).

Regards.

lelegard, 2019-05-03:
Hi,

There are two distinct topics: compiling with GCC 4.9 and compiling without PCSC or DTAPI (regardless of the compiler version). They are separately described in the page [Building the TSDuck binaries](https://tsduck.io/doxy/building.html). The second topic is described in section named _Building without specialized dependencies_. I want to keep them separate because they are independent. The user is then free to combine them as necessary.

The case of `NOTEST` is different because it _must_ be specified when compiling with GCC 4.9. There was a change in the C++ ABI for string handling between GCC 4.x and 6.x. So, C++ code which is compiled with GCC 4.x cannot be linked against C++ code from GCC 6.x when they both use strings. So, since the official compiler in the distro is GCC 6.x, the official CppUnit package is compiled with the same version. When you compile TSDuck with GCC 4.9, it cannot be linked against a CppUnit which was compiled with GCC 6.x. This is why `NOTEST=1` is required in that case. But it only excludes the unitary test program. It has no impact on the TSDuck binaries.

Note that the other dependencies such as PCSC or CURL have no problem since they are C libraries, not C++, and the C ABI did not change. DTAPI is a C++ library but it does not uses strings (except in one specific function that TSDuck carefully avoids to use for this reason) and is compatible between GCC 4 and 6.

lelegard, 2019-05-08:
Just one point

> I know, I know it! The list of compilers accepted for the last releases is reduced.

TSDuck does not "accept" or "reject" compilers. Some compilers are buggy and refuse to compile valid programs. The application does not reject the compiler, the compiler rejects itself from the field of correct programming!

The ones to blame are the distro makers (namely Debian), not GCC (the bug was later fixed). Packaging a known buggy compiler in a widely used and supported distro is not acceptable.

lars18th, 2019-05-10:
Hi @lelegard ,

I don't want to argue about it. You are quite right.
My suggestion is in the line of documenting which compilers are "supported", and which are "unsupported" for the reason X. It's frustrating for users to try to compile and find bugs.

In any case, suggesting to compile with `NOTEST=1 NOPCSC=1 NODTAPI=1` I feel it's good. Just to _simplify_ the dependencies to the minimum.
Don't you think that's a good idea?

lelegard, 2019-05-10:
Everything you mention is already documented in the [Building TSDuck](https://tsduck.io/doxy/building.html) page in the programmer's guide.

The restriction about GCC 6 and 7 and how to handle this in Debian is documented in the section "Specific GCC requirement on Raspian and Debian".

The build options to avoid PCSC, DTAPI, etc. are documented in the section "Building without specialized dependencies".

These two topics are independent. They can be used separately or jointly. It is up to the user to chose. I do not want to mix them because this is confusing. This would lead the user to believe that PCSC or DTAPI are not compatible with GCC 4.9, which is not the case. It is perfectly legitimate to use Dektec devices on any Linux versions, including Debian. Many users have adopted TSDuck because it supports Dektec devices. I do not want to publish a comment that would discourage them.

Consequently, unless some information is _missing_, I won't change the [Building TSDuck](https://tsduck.io/doxy/building.html) page.

lars18th, 2019-05-11:
> Consequently, unless some information is _missing_, I won't change the [Building TSDuck](https://tsduck.io/doxy/building.html) page.

OK. I agree. I close this issue.

closed 2019-05-11 by lars18th