tsduck/tsduck #280: Merge DVB's

alaaodeh, 2019-04-30:
Hi,

can we merge between two DVB's inputs like this and it will be stable or not ?

tsp -v --bitrate 60000000 -I dvb  --frequency 11919000000 --symbol-rate 27500000 --fec-inner 3/4 --polarity vertical -P merge "tsp -v --bitrate 60000000 -I dvb  --frequency 11623000000 --symbol-rate 27500000 --fec-inner 3/4 --polarity vertical"

rezna11, 2019-05-06:
No, not like this. You need `merge` for each DVB input as long as the first DVB input doesn't have enough null packets or unless you plan on removing most of the services to make that space. It's best to just use `null` input at a fixed bitrate, followed by regulate and 2x `merge`.
Note that you need 2 DVB tuners to do this and don't expect it to be stable. Merging in TSDuck isnt't meant for video/audio services.

alaaodeh, 2019-05-09:
Thanks for your kindly support,
how we can filter packets content please ,if we need to filter packet based on content from udp streaming input ?

lelegard, 2019-05-26:
> how we can filter packets content please ,if we need to filter packet based on content from udp streaming input ?

Using the latest option `--pattern` in plugin `filter`

closed 2019-06-04 by lelegard