tsduck/tsduck #281: Failed compilation: missed included in "tsdump.cpp"

lars18th, 2019-04-30:
Hi @lelegard ,

The master version has a bug related to a missing include that prevents to compile using GCC 4.9:

In this line https://github.com/tsduck/tsduck/blob/53063b35be048c2945a206a22bd1c72d1d178fda/src/tstools/tsdump.cpp#L241
The compilation fails because a missing `#include "tsUString.h"`:
```
  [LD] release-x86_64/tsdump
release-x86_64/tsdump.o: In function `MainCode(int, char**)':
tsdump.cpp:(.text+0x1d33): undefined reference to `ts::UString::toUTF8() const'
collect2: error: ld returned 1 exit status
../../Makefile.common:905: recipe for target 'release-x86_64/tsdump' failed
make[2]: *** [release-x86_64/tsdump] Error 1
```

So, please can you merge this patch?
```
diff --git a/src/tstools/tsdump.cpp b/src/tstools/tsdump.cpp
index 7f7e49fd..7793e90a 100644
--- a/src/tstools/tsdump.cpp
+++ b/src/tstools/tsdump.cpp
@@ -35,6 +35,7 @@
 #include "tsMain.h"
 #include "tsTSPacket.h"
 #include "tsPagerArgs.h"
+#include "tsUString.h"
 TSDUCK_SOURCE;
```

Regards.

lars18th, 2019-04-30:
Hi @lelegard ,

Merge the PR https://github.com/tsduck/tsduck/pull/282

lelegard, 2019-05-02:
I am sorry but this does not make sense for at least two reasons:

- The first line of tsdump.cpp includes tsMain.h. The first line of tsMain.h includes tsArgs.h. The first line of tsArgs.h includes tsReport.h. And, finally, the first line of tsReport.h includes tsUString.h. So, tsUString.h is in fact already included at the very beginning of tsdump.cpp. Including it again later is ignored because of its `#pragma once`.

- The undefined symbol error which is reported cannot come from a missing include. A missing include would have caused a compilation error.

Just to check, I just tried on a Raspbian 9.9 (stretch) on top of which gcc-4.9 and g++-4.9 have been installed. The build completes successfully.

I would assume that, at some point, your build failed or was interrupted for some reason. Or different versions of the compiler were used (first build with a version, then make without cleaning with another version of the compiler, rebuilding only a few modules). In that case, some inconsistencies may have appeared. And the solution was to recompile tsdump.cpp with the latest version of the compiler. So, the modification of the source file simply forced the recompilation, regardless of the useless additional include.

To validate this hypothesis, I suggest that you try again after reverting your modification (or checking out its previous commit):
~~~~
make distclean
make GCC=gcc-4.9 CXX=g++-4.9 AR=gcc-ar-4.9 NOTEST=true
~~~~

lars18th, 2019-05-03:
Hi @lelegard ,

Yes! Your assumption is right. After a clean checkout it compiles well.
So, I close it and I remove the PR.

Regards.

closed 2019-05-03 by lars18th