tsduck/tsduck #285: How edit the service name in tsduck

avinashmbd, 2019-05-02:


avinashmbd, 2019-05-02:
Hi ,
I am new to the tsduck software, So please can anyone help me how to edit and rename  the service in tsduck. Whenever I try to edit I am getting below error:

F:\Softwares\TSDuck-Win64-3.17-1189-Portable\TSDuck\bin>tsp -P svrename CT 1
tsp: svrename: too many parameter

F:\Softwares\TSDuck-Win64-3.17-1189-Portable\TSDuck\bin>tsp -P svrename -p CT 1
* Error: file: error reading file : Not enough storage is available to process this command. (8)

F:\Softwares\TSDuck-Win64-3.17-1189-Portable\TSDuck\bin>tsp -P svrename -p CT1
tsp: svrename: missing parameter

F:\Softwares\TSDuck-Win64-3.17-1189-Portable\TSDuck\bin>tsp -P svrename -p CT 1
* Error: file: error reading file : Not enough storage is available to process this command. (8)

F:\Softwares\TSDuck-Win64-3.17-1189-Portable\TSDuck\bin>tsp -P svrename -p CT 1
* Error: file: error reading file : Not enough storage is available to process this command. (8)

F:\Softwares\TSDuck-Win64-3.17-1189-Portable\TSDuck\bin>-P sdt -s 1 --cleanup-private-descriptors --provider "test"
'-P' is not recognized as an internal or external command,
operable program or batch file.

F:\Softwares\TSDuck-Win64-3.17-1189-Portable\TSDuck\bin>tsp -P sdt -s 1 --cleanup-private-descriptors --provider "test"
* Error: file: error reading file : Not enough storage is available to process this command. (8)

Thanks,
Avinash

lelegard, 2019-05-02:
Please read the TSDuck documentation to understand the _too many parameters_ or _missing parameter_ error messages.

avinashmbd, 2019-05-03:
Hi,
Can someone help **how to change the lcn number.** I tried below all the possible commands but it's not changing. I am able to change the service name but **lcn number is not changing**

tsp -I file input.ts -P svrename XYZ -l 71  -n "test2" -O file output.ts   - Using service name
tsp -I file input.ts -P svrename 0x0468 -l 71  -n "test2" -O file output.ts  - Using service ID
tsp -I file input.ts -P svrename 58 -l 71  -n "test2" -O file output.ts  - Using LCN number
tsp -I file input.ts -P svrename xyz -l "40" -O file output.ts  - Using service name

Please can anyone help me out on this.

lelegard, 2019-05-03:
Duplicate of #286

lelegard, 2019-05-10:
Closed as duplicate

closed 2019-05-10 by lelegard