tsduck/tsduck #286: How to change the LCN number

avinashmbd, 2019-05-03:
Hi,
Can someone help me out how to change the lcn number. I tried below all the possible commands but it's not changing. I am able to change the service name but lcn number is not changing

tsp -I file input.ts -P svrename XYZ -l 71 -n "test2" -O file output.ts - Using service name
tsp -I file input.ts -P svrename 0x0468 -l 71 -n "test2" -O file output.ts - Using service ID
tsp -I file input.ts -P svrename 58 -l 71 -n "test2" -O file output.ts - Using LCN number
tsp -I file input.ts -P svrename xyz -l "40" -O file output.ts - Using service name

Please can anyone help me out on this.

lelegard, 2019-05-03:
What do you mean by "it's not changing"? How do you see the output LCN values?

The LCN is unfortunately defined by no standard. The LCN can be defined in a number of different ways, using private descriptors. Sometimes, it is not even defined at all. The most common way to define it is to use the EACEM-defined private descriptor named _logical_channel_number_descriptor_. This is what the plugin `svrename` is processing.

So, if the input file contains a NIT with EACEM-defined _logical_channel_number_descriptors_, then the rename operation will modify them with the new LCN value. If any other operator-specific method is used to specify LCN values, then you cannot change them.

To check the content of the NIT, please post the output of:

~~~
tstables input.ts -p 0x10 -t 0x40
~~~

avinashmbd, 2019-05-07:
> What do you mean by "it's not changing"? How do you see the output LCN values?
>
> The LCN is unfortunately defined by no standard. The LCN can be defined in a number of different ways, using private [...]
Please find the NIT dats:

Descriptor 1: unknown (0x83, 131), 96 bytes
      0000:  04 5C 80 41 03 F5 80 15 03 F9 80 17 05 1E  .\.A..........
      000E:  80 74 04 98 80 70 05 66 80 1A 05 4E 80 6A  .t...p.f...N.j
      001C:  05 84 80 72 05 62 80 73 04 6E 80 2D 04 5E  ...r.b.s.n.-.^
      002A:  80 2C 04 1F 80 49 04 05 80 51 04 06 80 46  .,...I...Q...F
      0038:  04 10 80 4A 04 20 80 47 05 2C 80 80 05 6A  ...J. .G.,...j
      0046:  80 4C 04 28 80 1C 05 18 80 71 04 99 80 6F  .L.(.....q...o
      0054:  04 2A 80 75 04 D0 80 76 05 28 80 86        .*.u...v.(..
    - Descriptor 2: Satellite Delivery System (0x43, 67), 11 bytes
      Orbital position: 75.0 degree, east
      Frequency: 12.16000 GHz
      Symbol rate: 45.0000 Msymbol/s
      Polarization: linear - vertical
      Modulation: DVB-S2, 8PSK, alpha=0.35.

Please help me out on this...

lelegard, 2019-05-07:
Please post the complete NIT or (preferably) attach the binary NIT.

The descriptor 0x83 is probably a private LCN descriptor. But it appears as "unknown". So there is no preceding private data specifier (PDS). We can be only sure after getting the complete NIT.

The most probable cause (and I have already seen that) is that the operator's signalisation is incorrect. In DVB sytems, it is incorrect to have a private descriptor (> 0x7F) without being preceded by the appropriate PDS descriptor. There are several different private descriptors with tag 0x83 and at least two of them are different implementations of an LCN descriptor (one is defined by EACEM and one is defined by Eutelsat). So, in a conformant DVB system, descriptor 0x83 without preceding PDS descriptor is undefined.

But some operators use such incorrect signalisation when their own STB's are coded to directly interpret private descriptors their own way, without checking PDS descriptors. In short, an operator uses incorrect signalisation with incorrect STB's, making a working but non-standard system.

If this is the case, no standard tool such as TSDuck can properly handle the LCN. But we need to get the full NIT to make sure that this hypothesis is correct.

closed 2019-06-03 by lelegard