tsduck/tsduck #289: Plugin filter: missing functionality

lars18th, 2019-05-08:
Hi @lelegard ,

With the new added support of "labels" I lack some funtionality with the `filter` plugin:

- `--unset-label`: This is equivalent to `--set-label` but instead of "set" it "clears" the mark.
- `--set-label-false`: This is the action of tagging when the condition is false..
- `--unset-label-false`: The mix of the two above.

Without this it's impossible to "clear" marks. And set/clear for the second part of the test I feel is relevant.

Example: You like to unset a flag when a condition is false and the input has `--only-label`. In this case the parameter `--unset-label-false` has a full sense.

You agree?

lelegard, 2019-05-08:
OK for `--unset-label`.

>--set-label-false: This is the action of tagging when the condition is false..
--unset-label-false: The mix of the two above.

What about `--negate --set-label` and `--negate --unset-label` ?

lelegard, 2019-05-08:
Option `--reset-label` (not "unset") added.

lars18th, 2019-05-10:
> > --set-label-false: This is the action of tagging when the condition is false..
> > --unset-label-false: The mix of the two above.
> 
> What about `--negate --set-label` and `--negate --unset-label` ?

Becuase my suggestion simplifies the use of the plugin and it doesn't interfere with the current logic. Using the alternative of `--negate` implies in some cases to use **two** plugins.

The idea is instead of:
```
    if (ok) {
        pkt_data.setLabels(set_labels);
        pkt_data.clearLabels(reset_labels);
        return TSP_OK;
    }
    else {
        return drop_status;
    }
```

Something like:
```
    if (ok) {
        pkt_data.setLabels(set_true_labels);
        pkt_data.clearLabels(reset_true_labels);
        return TSP_OK;
    }
    else {
        pkt_data.setLabels(set_false_labels);
        pkt_data.clearLabels(reset_false_labels);
        return drop_status;
    }
```

And then we can use this syntax:
`-P plugin --only-label 1 -n --payload --pcr --set-label 2 --set-label-false 3`

Remember that the "OR" condition is implicit, so getting a control over the TRUE and FALSE can be desirable. Why? Because the negation of and OR it's the AND of the negation of the two elements.
You agree?

lelegard, 2019-05-10:
> Becuase my suggestion simplifies the use of the plugin and it doesn't interfere with the current logic. Using the alternative of --negate implies in some cases to use two plugins.

I remember that in some old issue I mentioned the fact that the KISS principle implies that it is useless to complexifiy a plugin if the same thing can be done using two simpler plugins.

lars18th, 2019-05-11:
> I remember that in some old issue I mentioned the fact that the KISS principle implies that it is useless to complexifiy a plugin if the same thing can be done using two simpler plugins.

As usual, you're absolutely right. I appreciate and respect your experience very much.
I close this issue then, since all the functionality is already implemented. :+1:

closed 2019-05-11 by lars18th