tsduck/tsduck #290: Plugin remap: unconditional remapping

lars18th, 2019-05-08:
Hi @lelegard ,

With the labelling support I miss this functionality:

- `-P remap --only-label 1 -n -u *=100`

The idea is quite simple: remap to pid 100 _any_ incomming packet. I suggest this syntax as I feel is more clear to implement it. However, if you prefer another one...

lars18th, 2019-05-08:
Hi @lelegard ,

If you will implement my request #291 about the `patcher` plugin (or the modification of current `craft` input) then this request is innecessary. The other plugin will have the option `--pid` and then this request can be discarded.

Please, consider the best option.
Regards.

lelegard, 2019-05-08:
Hi @lars18th 

> If you will implement my request #291 about the patcher plugin (or the modification of current craft input) then this request is innecessary. The other plugin will have the option --pid and then this request can be discarded.

I will go this way. I consequently close this one.

closed 2019-05-08 by lelegard