tsduck/tsduck #291: plugin "patcher"

lars18th, 2019-05-08:
I feel the same functionality of the `craft` plugin can be reused for a plugin `patcher`. With the new label support, it has sense to apply _all_ (or nearly all) parameters from `craft` to valid input packets using `--only-label`.

Or perhaps you prefer to extend the current `craft` plugin to be not only "input" but "plugin" too.

In any case, set/clear flags/values like "pusi", "es-priority", "error", "discontinuity", etc is very valuable.

I hope you like to consider it.
Regards.

lelegard, 2019-05-08:
OK, good idea. There will be some limitations though. Some stuff such as "discontinuity" are stored in the adaptation field. If an existing packet has no adaptation field, creating one would destroy part of the existing payload. I am not sure that this would be a responsible behaviour.

lars18th, 2019-05-10:
> I am not sure that this would be a responsible behaviour.

If you want to be polite then add the `-u|--unconditional` parameter. Without it (by default) _all_ transformations should be verified for validation. But with it, all the controls are deactivated. The responsibility then lies with the user.

You agree?

lelegard, 2019-05-10:
> If you want to be polite then add the -u|--unconditional parameter. 

OK, but then what should we do if "discontinuity" shall be set on a packet without adaptation field? Overwrite the start of the payload with the AF? Shift the payload and truncate it to make room for the AF? Remove the payload completely? And what should we do with the PUSI? If it is set, this means that the start of the payload means something. And, no, I do not want another option to specify which form of corruption (because the packet payload becomes useless anyway) to apply on the packet.

This is why I prefer to not apply a modification if it should be set in the AF and there is no AF or the AF is too short to fit.

lars18th, 2019-05-10:
Please, don't try to target _all_ scenarios. This isn't my suggestion (I'm sorry if I implied that).

Just do that: If some action it's impossible (or very complex) then write a warning and pass the packet. :peace_symbol:

lars18th, 2019-05-13:
Hi @lelegard ,

I see the new implementation of the `craft` plugin. I'm only reviewing the code, and as regular it's a good implementation. :+1: 

However, I feel that I see the lack some small functionality: `--payload-size -1`. The semantic in this case it's this: "resize to the size of the current payload". And this has sense? Yes, when the payload is a PES packet that have stuffing bytes in the PES header (remember my request at #273). If you implement it in this way (-1 = remove the stuffing from the PES header) then maybe it's better.

Do you agree?

lelegard, 2019-05-14:
I'd rather prefer a specific option such as `--pack-pes-header`.

Please provide a real-world TS file with such a packet to pack (one packet is enough).

lars18th, 2019-05-14:
> I'd rather prefer a specific option such as `--pack-pes-header`.
> 
> Please provide a real-world TS file with such a packet to pack (one packet is enough).

It's here: https://github.com/tsduck/tsduck/issues/273#issuecomment-490535945

lelegard, 2019-05-14:
Please attach the binary file. I want to add it in the test suite and I need a TS file.

lars18th, 2019-05-14:
> Please attach the binary file. I want to add it in the test suite and I need a TS file.

OK. Here it is.

[data_packet.zip](https://github.com/tsduck/tsduck/files/3176288/data_packet.zip)

lars18th, 2019-05-14:
Hi @lelegard ,

This is just one packet. You prefer the three packets of the entire PES?

lars18th, 2019-05-14:
Hi @lelegard ,

Here the entire PES packet of this Teletext page (total of 4 TS packets):
[teletext_PES_packet.zip](https://github.com/tsduck/tsduck/files/3176366/teletext_PES_packet.zip)

I feel this will be more useful

lelegard, 2019-05-14:
Hi @lars18th 

I have added option `--pack-pes-header` in the `craft` plugin. It seems to work with your file. Please do more tests. Also note that I had no time to fully test `craft` as a packet processor plugin. So, it's your turn to do the testing job 😉

lars18th, 2019-05-15:
> I have added option `--pack-pes-header` in the `craft` plugin. It seems to work with your file. Please do more tests. Also note that I had no time to fully test `craft` as a packet processor plugin. So, it's your turn to do the testing job

OK @lelegard ,

However, please commit it. I don't see this `--pack-pes-header` in the `craft` plugin of the current tree. :wink:

lelegard, 2019-05-15:
I committed but I forgot to push, sorry 😞 
This is done now.

lars18th, 2019-05-15:
OK. I'll try tomorrow.

lelegard, 2019-05-16:
There was a bug in the handling of PES header flags and the PTS was deleted. It is fixed now.

lars18th, 2019-05-21:
Hi @lelegard ,

Thank you for the `--pack-pes-header` parameter. I'm doing a lot of tests, but I need more time to complete them. I'll close this issue when I can confirm that it's working well.

Regards.

lars18th, 2019-05-23:
After some tests the parameter `--pack-pes-header` seens to work as expected.
So I close this issue.

closed 2019-05-23 by lars18th