tsduck/tsduck #292: AsyncReport crash

nhjm449, 2019-05-08:
First: Thank you for the time and effort you put into this project. It's quite impressive. :)

`tsp` occasionally crashes under heavy logging due to what appears to be a race condition.

v3.16-1110, compiled in debug mode
GCC 7.3.0 (Arch Linux) and GCC 4.8.5 (CentOS 7)

```
$ tsp -I null 900000 -P count -a -O file /dev/null
[...count logs...]
pure virtual method called
terminate called without an active exception
Aborted (core dumped)
```

(Note: Specifying the count filter multiple times seems to increase the chance of failure.)

Backtraces point to `AsyncReport` and `SafePtr`... where using `NullMutex` seems dangerous.

https://gist.github.com/nhjm449/f8a78a8fcf44830e55705885db306d8d
https://gist.github.com/nhjm449/f10bde988619bc184823cff54df1038f

I've also observed a `std::bad_alloc` exception in release mode that smells like the same root cause, it's just harder to trigger.

lelegard, 2019-05-09:
Hi @nhjm449,

I cannot reproduce this. I tried on a VM with CentOS 7.6 and GCC 4.8.5 and it never crashes. I used the latest code from the repo, not v3.16-1110. But, from the backtrace you provided, there was not much change in the code surrounded the crash. However, if there is a race condition, the hardware or VM type, the number of CPU, the system load, etc. may have an impact and make a problem reproduceable or not.

> Backtraces point to AsyncReport and SafePtr... where using NullMutex seems dangerous.

This was intentional but looking very closely, it is probably incorrect.

The mutex in SafePtr is used to protect the dereferencing of the pointer and the management of its internal reference counter. In AsyncReport, a message is first handled in a thread (typically a plugin thread) and then passed to the report thread using a message queue. The message queue has an internal mutex (not a null one). So, the component which is potentially used in concurrent threads is the message queue and it is properly synchronized. The objects which are passed by the queue are the SafePtr. And these objects are, in theory, first used in one thread _and then_ the other thread. So, they do not need to be synchronized. However, inspecting the code closely, some SafePtr variables which contained the safe pointer before passing the message are destructed after having passed the message to the other threads. And here we may have a race condition. This is consistent with the message "pure virtual method called" which is usually a symptom of using a partially destructed object. This is also consistent with an `std::bad_alloc` exception if there is a double deallocation because of insufficient synchronization.

It seems amazing to me that I made such a mistake. This is normally the kind of thing I take great care of. This is likely the root cause of your problem.

I will fix this and push the fix. I will let you know when you can test it. However, this will be on the latest development version v3.18, not v3.16. This will be OK on CentOS with GCC 4.8.5. This will be also OK with recent Ubuntu or Fedora with GCC 8.x or 9.x.

However, this probably won't compile with GCC 7.3. As reported in other issues here, GCC 6.x and 7.x are broken and refuse to compile valid C++ code in TSDuck which was introduced in v3.17. This bug was fixed into GCC 8.x at least (I do not know which precise version fixed that bug). I do not blame GCC. Bugs happen and they were properly fixed. But I do blame distro makers (Debian and apparently ArchLinux) who release current systems with old broken compilers. If you want to use Debian, Raspbian, ArchLinux, then you have to install either GCC 4.9 or GCC 8.x as indicated [here](https://tsduck.io/doxy/building.html#reqraspbian). I do that on my Raspberry Pi using Raspbian. Because of this bug (or any other reason), the Debian and Raspbian repos contain GCC 4.9 and simply use `apt install` to get them. I do not know about ArchLinux, I never used it. My Linux targets are Fedora, Ubuntu, Raspbian and CentOS only.

lelegard, 2019-05-10:
Hi @nhjm449 

The fix in in commit 45ccf69618924fb5e6995029c86842bcb5b07f18.

Since the problem occurs in your environment but not mine, please pull the repo, rebuild and test to check if this is fixed and let us now the result. Take care of the buggy GCC versions as described in the previous post. The easiest way is using your CentOS since GC 4.8.5 is fine.

nhjm449, 2019-05-10:
Thanks!

Naturally, I haven't been able to reproduce the issue today on the CentOS machine today. But I can say that the new version hasn't crashed so far.
But I can confirm that the (backported) fix seems to fix the problem on the other machine. :smiley:

(Arch Linux is a bleeding-edge, rolling release distro, so the fact that I'm still on GCC 7.x is solely my fault as I haven't run a system update in a while. And I'm tied to the v3.16 plugin API for now.)

closed 2019-05-11 by lelegard