tsduck/tsduck #294: Plugin filter: Send signal

lars18th, 2019-05-11:
Hi @lelegard ,

This isn't the same as commented in #289 , but it _can_ be related with the `filter` plugin:

- When using flags it can be interesting to "send" a signal when a flag appears. This "send" message is in the form of the UDP datagrams used by other filters. And as the `filter` plugin can "trigger" actions it seems like a reasonable candidate.

So I suggest (if you agree) to add this new parameter:

- `-P filter --send-to [udp:port:{string}] [mode]`: So when the test is true (like when using `--label`) one UDP datagram it's sended to the destination with the {string} message. Futhermore, the `mode` is a selector of the behaviour.
- Mode 0: Send a datagram just only one time (the default).
- Mode 1: Send a datagram every time the condition goes true (this implies to save the status in the filter). That's when the flag appears.
- Mode 2: Send a datagram every time the condition goes false. That's when the flag disappears.
- Mode 3: Send a datagram every time (potentially dangerous because the overhead).

And what I'm not sure it's about a "number of times" or "distance" related to the modes 1 and 2. Perhaps, it has sense to apply some "delay" or "offset" after the condition. For example, if you get 0,0,0,0,1,1,1,0,1,1,0,0,0,0 and the distance is 2, then only 1 message is sended: after the first 1, as after just the only one 0 between the two 1 the distance is less than 2.

What you think about this idea?

lars18th, 2019-05-11:
Hi @lelegard ,

> And what I'm not sure it's about a "number of times" or "distance" related to the modes 1 and 2. Perhaps, it has sense to apply some "delay" or "offset" after the condition. For example, if you get [...]

Regarding this part, I have an additional idea: add these two new parameters too:
- `--mute-packets [number]`: This "disconnects" the check (globally, that's for all options) for the "number" of incomming packets (in this case, the incomming is all packets that will be **checked**, and not the total). 
- `--mute-time [seconds]`: Instead of a count, this is timeout for disconnecting the check.

With these new parameter will be possible to apply checks only inside intervals for any check. Futhermore the complez case described above can be targeted too. And they complement also the already implemented `--every`.

As a summary, add to the `filter` plugin:
1. `--mute-packet` and `--mute-time` to complement `--every`.
2. `--send-to` with the four described semantics to send messages to other processes.

I think this change will open up many new options with a small effort.
You agree?

lelegard, 2019-05-11:
Well, well, well. So many things are possible. During the first ten years of TSDuck, before it became an open source project, I implemented features that I needed or that my close colleagues needed. Since it's open source, most improvements were made because some users (me included) needed them. TSDuck is perceived as useful because everything it can do was actually needed at some point.

Does any one really _need_ this? Or would it just be a nice to have?

lars18th, 2019-05-13:
> Does any one really _need_ this? Or would it just be a nice to have?

Well, I'll try to discuss it (the `--send-to` parameter):

- Using TSDuck as a Library it's useful for some type of projects. However, for _prototyping_ it's better to use the `tsp.exe` approach. Or almost that's my point of view. However, when using the executable it's quite difficult to "connect" the TSDuck process with other processes. Yes, you can use pipes, but this is mainly to share the stream, and not to share "signals". In this way, enhancing the `filter` plugin with the functionality of sending messages to outside (using simple UDP messages) it's possible to "announce" to anyone when a new LABEL appears. In this way after the implementation of the label functionality it's now possible to enhance other plugins to set/unset these flags (I'm thinking about one or two, but now are only ideas and not requirements). But the key point is to control the `tsp.exe` chain from external tools. For example, you can execute a broker and receive/send messages from/to the TSDuck process. Then it will be very easy to manage the worflow using external processes.

Perhaps this scenario goes beyond the current goals of the TSDuck. But in my opinion, as other plugins already support the UDP communication, then it might be easy to add it to `filter` as well.

Anyway, I'm just talking about usability. And not forcing you to implement something irrelevant. At this point it's enough that you can comment on it.

- In the other hand, the `--mute-packet` and `--mute-time` parameters can be used as a good complement for the current functionality. Imagine as example, that you want to "trigger" the insertion of new tables when some event appears in the stream. At time you can use a flag and the `filter` plugin. But in this case you lack the functionality to enforce some restriction like: not repeat it until X seconds, or wait until... I'm sure you understand the relevance of these two "mute" parameters.

Please, comment it.
Regards.

lelegard, 2019-05-25:
Hi @lars18th 

The new plugin `trigger` should do what you want, more or less. Please review and test it.

lars18th, 2019-05-25:
> The new plugin `trigger` should do what you want, more or less. Please review and test it.

For sure I'll do it. Thank you! :+1:

closed 2019-07-04 by lelegard