tsduck/tsduck #296: filter  specific packet

michaeladams123, 2019-05-12:
Hi,

how to filter specific packet PID  from streaming ?

Regards

lelegard, 2019-05-12:
I suggest that you start by reading the TSDuck user's guide which is installed with the binaries or is available online [here](https://tsduck.io/download/docs/tsduck.pdf). Read the command and plugin reference chapters and the examples chapter. Most questions are answered here.

First, define the word "streaming". If you mean IPTV multicast, then use input plugin `ip`. If you mean HLS, then use input plugin `hls`. Then, to filter a specific PID use plugin `filter`.

michaeladams123, 2019-05-13:
I already checked on documents but I need to detect a specific but from packets from IPTV multicast

like this but not working

    Provide a "check" plugin that finds if a packet has some values (for example 0x81;0x11;0x00 at given offset) and then it executes an action (in this case change pid). One example:
    -P check-packet -pid 100 --label 0x81;0x11;0x00 --label-offset 35 --change-pid 101

And the options can be --change-pid, --remove, --duplicate, --patch (where "patch" is like "label" and "label-offset" to change content of the packet).

lelegard, 2019-05-13:
There currently no such plugin, unfortunately.

michaeladams123, 2019-05-13:
when will be available please ?, or any workaround solution

lars18th, 2019-05-13:
Hi @michaeladams123 ,

I feel I undertand your request, and this can be the syntax to achieve your goal:

```
tsp \
 -I ip 224.0.0.1:1234 \
 -P filter -p 100 --set-label 1 \
 -P filter --only-label 1 --pattern 0x81;0x11;0x00 --pattern-offset 35 --set-label 2 \
 -P craft  --only-label 2 --pid 101 \
 -O ip 224.0.0.2:1234
```

The input multicast will be 224.0.0.1:1234 and the processed output multicast will be in 224.0.0.2:1234.

This is not yet possible. First because the `craft` plugin is on-road (however, it's in beta). But the second reason it's because the `filter` plugin doesn't have the parameters `--pattern` and `--pattern-offset`.

However, it is possible that @lelegard will agree to put it into practice.
Regards.

lelegard, 2019-05-13:
Options `--pattern` and `--pattern-offset` could probably be added to plugin `filter` with the following syntax:
~~~
 -P filter --only-label 1 --pattern 811100 --pattern-offset 35 --set-label 2
~~~
This is more concise and using semi-colons on a command line is a pain on Unix systems.

All current options in TSDuck which use a binary pattern of variable size use that compact hex-only syntax.

lelegard, 2019-05-16:
Options `--pattern`, `--search-payload` and `--search-offset` have been added to plugin `filter`.

closed 2019-05-24 by lelegard