tsduck/tsduck #297: Cannot build on Debian 4.9.168-1

nt74, 2019-05-13:
I have cloned the latest git and when trying to build with the command "make deb" i get [errors.]

[debian_build_issues.txt](https://github.com/tsduck/tsduck/files/3171604/debian_build_issues.txt)

lelegard, 2019-05-13:
Hi @nt74,

Similar issues were already reported in #279, #269, #231 and #219. The latest version of Debian includes GCC 7.x. This compiler is buggy, broken. All known versions of GCC 6.x and 7.x have this bug. The error you see is the compiler rejecting valid C++ code. TSDuck can be compiled with GCC up to 4.9 and then 8.x and 9.x.

The latest versions of all popular distros, _except Debian_, have valid compilers:
- Red Hat EL, CentOS: GCC 4.8.5
- Ubuntu: GCC 8.x
- Fedora: GCC 9.x

It is a shame that a reputed distro such as Debian continues to ship a broken & buggy compiler with the latest version. Bugs happens but shall be fixed. The GCC team fixed the bug in version 8.x and higher. The Debian team did not make a proper job by not upgrading to the fixed version.

You can workaround the problem on Debian either by downgrading GCC to 4.9 (available from the official repo) or by upgrading to 8.x (third-party packages available). This is documented in the TSDuck programmer"s guide in page "[Building TSDuck on Debian and Raspbian](https://tsduck.io/doxy/building.html#reqraspbian)".

lars18th, 2019-05-13:
Hi @lelegard ,

I completely agree with what you're saying. :100: 

However, perhaps it's time to include in the sources a check of the GCC version used to compile. And then print an ERROR WARNING aborting the compilation if the compiler used it's GCC 6.x or 7.0. In this message you can include the URL of the documentation "Building TSDuck on Debian and Raspbian".

Here how to implement it in "tsduck.h" for example:
https://gcc.gnu.org/onlinedocs/cpp/Common-Predefined-Macros.html

lelegard, 2019-05-15:
>However, perhaps it's time to include in the sources a check of the GCC version used to compile

This is now done, compilation should fail with an explicit error message when compiling with GCC 6 and 7.

lars18th, 2019-05-15:
Hi @lelegard ,

> This is now done, compilation should fail with an explicit error message when compiling with GCC 6 and 7.

OK. Tested and working... :+1: 
But I need to comment two questions before closing this issue:

1. After this change, the compilation fails when using the GCC-4.9 backport. The correct command needs to be `make CC=gcc-4.9 GCC=gcc-4.9 CXX=g++-4.9 AR=gcc-ar-4.9 NOTEST=true`. This is **required** because in the `Makefile.common` the `$(CC)` variable it's already used. I don't know why mixing GCC 6.x for C and GCC 4.9 for C++ the compilation works.
2. I recommend to update the documentation at https://tsduck.io/doxy/building.html#reqraspbian adding this information: To install the GCC 4.9 compiler in a Debian 9.x Stretch you need to add the line `deb http://http.debian.net/debian oldstable main contrib non-free` inside the file `/etc/apt/sources.list` and then execute `apt-get update; apt-get install gcc-4.9 g++-4.9`. That's all required to install a compatible compiler in Debian/Ubuntu.

I hope you can update the documentation soon.
Regards.

lelegard, 2019-05-16:
Doc updated (in the repo, not yet on the web site)

lars18th, 2019-05-16:
LGTM,
So please, close this issue.

closed 2019-05-16 by lelegard