tsduck/tsduck #298: Compiler errors: Raspbian 9.9, GCC 9.1.0

alexander-nf, 2019-05-13:
Hello Thierry!

I haven't been using the toolkit for long and started to build TSDuck 3.18-1245 for the [Raspberry PI TV HAT](https://www.raspberrypi.org/products/raspberry-pi-tv-hat/) module testing.

First of all, I got compilation errors after `make`-ing the build by GCC 6.3.0. I found information about "[Specific GCC requirement on Raspian and Debian](https://tsduck.io/doxy/building.html)", followed the external link provided there and read the article with the instructions how to install a newer GCC version on Raspbian.

1. It seems [the article](https://solarianprogrammer.com/2017/12/08/raspberry-pi-raspbian-install-gcc-compile-cpp-17-programs/) was updated by the author recently, because it describes an installation of GCC **9.1.0** but https://tsduck.io/doxy/building.html says the article "_explains how to install GCC **8.1.0** on Raspbian_".

2. I have installed GCC 9.1.0, ran `make` and got an error:
```
tsThread.cpp: In member function 'bool ts::Thread::start()':
tsThread.cpp:207:54: error: 'PTHREAD_STACK_MIN' was not declared in this scope
  207 |         const size_t size = RoundUp(std::max<size_t>(PTHREAD_STACK_MIN, _attributes._stackSize), SysInfo::Instance()->memoryPageSize());
      |                                                      ^~~~~~~~~~~~~~~~~
../../Makefile.common:898: recipe for target 'release-arm/tsThread.o' failed
make[2]: *** [release-arm/tsThread.o] Error 1
```
I changed the `PTHREAD_STACK_MIN` to `0` and this kludge gave me ability to go further.

3. The second compiler error arose on the utests stage:
```
[CXX] utestAlgorithm.cpp
utestAlgorithm.cpp: In constructor 'AlgorithmTest::CppUnitExDeleter::CppUnitExDeleter()':
utestAlgorithm.cpp:61:5: error: zero as null pointer constant [-Werror=zero-as-null-pointer-constant]
   61 |     CPPUNIT_TEST_SUITE_END ();
      |     ^~~~~~~~~~~~~~~~~~~~~~
cc1plus: all warnings being treated as errors
../../Makefile.common:898: recipe for target 'release-arm/utestAlgorithm.o' failed
make[2]: *** [release-arm/utestAlgorithm.o] Error 1
```
I commented `-Wzero-as-null-pointer-constant` in `CXXFLAGS_WARNINGS` of Makefile.common and successfuly compiled the project. I'm sure I did the dirty things but I have no skill to say what corrections should be done and where.

Could you please comment points 2 and 3? I haven't found information in similar or closed [issues](https://github.com/tsduck/tsduck/issues). Am I totally wrong or something should be changed in the project code, the compiler instructions, somewhere else?

lars18th, 2019-05-13:
Hi @alexander-nf ,

First of all, try to compile with `make NOTEST=true` and not the plain `make`. Futhermore, in my compilations _every_ time I use `make NOTEST=1 NOPCSC=1 NODTAPI=1`. This will solve for sure your _second_ error.

Regarding the first one, just try to install the stock 4.9 compiler. The current version of TSDuck can't be compiled with GCC 6.x or 7.x. But it could be with the 4.8 & 4.9. I recommend it as it is much easier to install than the vanilla version of GCC 9.1.0.

See more info about this problem in this issue #279.

Regards.

alexander-nf, 2019-05-14:
Hi @lars18th!
Thank you for the explanations.

I've read the Doxygen manual and the origin (#279). Also I have successfully compiled the project using `make GCC=gcc-4.9 CXX=g++-4.9 AR=gcc-ar-4.9 NOTEST=true`.

But, as I understand from the manual (and from the [comment](https://github.com/tsduck/tsduck/issues/297#issuecomment-491719007)), the GCC 9.x should compile the project without an issue and additional options of `make`. For this reason I reported the problem. If I am wrong or passed something by, I'm sorry and it is possible to close this issue thread as a duplicate.

Yes, I also confirm that `NOTEST=true` helps to avoid the point 3 from my initial message and to compile the TSDuck with GCC 9.1.0 successfully. But I still can't get rid of the GCC 9.1.0 `error: 'PTHREAD_STACK_MIN' was not declared in this scope`.

lars18th, 2019-05-14:
Hi @alexander-nf ,

I'm glad you solved your compilation problems.

Regarding the use of the GCC 9.x in Debian and Raspian, from my experience: **forget it at all!** It's quite difficult to mix the stock compiler with a new vanilla compiler inside the same distribution, except if you use the "cross-compiling" approach. For this reason I suggest to install a failback compiler, like the GCC 4.9 that you can found on these distributions.

In any  case I hope that @lelegard will include soon an error message when compiling with GCC 6.x & 7.0 as commented here: https://github.com/tsduck/tsduck/issues/297#issuecomment-491894281

Regards.

lelegard, 2019-05-15:
> In any case I hope that @lelegard will include soon an error message when compiling with GCC 6.x & 7.0 as commented here: #297 (comment)

This is now done, compilation should fail with an explicit error message when compiling with GCC 6 and 7.

lars18th, 2019-05-15:
Please, close this issue. But read first my comment at https://github.com/tsduck/tsduck/issues/297#issuecomment-492794506 :wink:

alexander-nf, 2019-05-16:
Hello,

I think if there were no any option to use GCC 8.1+ in "[Specific GCC requirement on Raspian and Debian](https://tsduck.io/doxy/building.html#reqraspbian)" I've never opened the issue. But it's OK, according to your and @lelegard's recommendations I've chosen to use GCC 4.9 and to stop experiments of integrating GCC 9.1.0 in Raspbian. The experiments could take place, but it isn't in the scope of the project and the main target to build the project is achievable with GCC 4.9.

@lars18th and @lelegard,
thank you for your attention and patience.

closed 2019-05-16 by alexander-nf