tsduck/tsduck #299: tables plugin - problem with duplicate transport stream id

milos-sedlacek, 2019-05-14:
When extracting data from SDTo table containing multiple loops of same transport_stream_id, the information about these transport streams are appended endlessly to output files. 
I use multicast as input.
`$ tsp -I ip 239.170.13.1:10001 -P tables -t 70 --xml-output out_xml_SDTo.xml -O drop`

The XML output looks like this
`<SDT version="27" current="true" transport_stream_id="0x0005" original_network_id="0x0001" actual="false">
    
    <service service_id="0x0D20" EIT_schedule="true" EIT_present_following="true" CA_mode="true" running_status="running">
      <service_descriptor service_type="0x01" service_provider_name="" service_name="Duna"/>
    </service>
  </SDT>
  <SDT version="10" current="true" transport_stream_id="0x0003" original_network_id="0x1111" actual="false">
    
    <service service_id="0x0BCF" EIT_schedule="true" EIT_present_following="true" CA_mode="true" running_status="running">
      <service_descriptor service_type="0x19" service_provider_name="" service_name="Mooz Dance HD"/>
    </service>
  </SDT>
  <SDT version="11" current="true" transport_stream_id="0x0005" original_network_id="0x1111" actual="false">
    
    <service service_id="0x03F3" EIT_schedule="false" EIT_present_following="false" CA_mode="true" running_status="running">
      <service_descriptor service_type="0x01" service_provider_name="" service_name="Filmbox HU"/>
    </service>
    <service service_id="0x03F8" EIT_schedule="true" EIT_present_following="true" CA_mode="true" running_status="running">
      <service_descriptor service_type="0x01" service_provider_name="Slovak Telekom" service_name="Publika TV"/>
    </service>
    <service service_id="0x03F9" EIT_schedule="true" EIT_present_following="true" CA_mode="true" running_status="running">
      <service_descriptor service_type="0x01" service_provider_name="Slovak Telekom" service_name="Credo TV"/>
    </service>
  </SDT>
  <SDT version="30" current="true" transport_stream_id="0x0003" original_network_id="0x0001" actual="false">
    
    <service service_id="0x04A6" EIT_schedule="true" EIT_present_following="true" CA_mode="true" running_status="running">
      <service_descriptor service_type="0x01" service_provider_name="" service_name="Hir TV"/>
    </service>
    <service service_id="0x0A96" EIT_schedule="true" EIT_present_following="true" CA_mode="true" running_status="running">
      <service_descriptor service_type="0x01" service_provider_name="" service_name="Galaxy 4"/>
    </service>
    <service service_id="0x0CEE" EIT_schedule="true" EIT_present_following="true" CA_mode="true" running_status="running">
      <service_descriptor service_type="0x01" service_provider_name="RCS" service_name="Film 4"/>
    </service>
  </SDT>`
... and continues

lelegard, 2019-05-14:
Sorry for that. Could you please post the binary file of the SDT (from ` -P tables -t 70 --bin sdt.bin`) ?

milos-sedlacek, 2019-05-17:
No problem. I can overcome the problem by reading the duplicate records just once. I wanted to make you aware of this. 
sdt.bin attached. Unzip first.
[sdt.zip](https://github.com/tsduck/tsduck/files/3190496/sdt.zip)

lelegard, 2019-05-17:
Hello,

From your original post, I had the impression that the `tables` plugin was looping forever, which would have been a bug of course. Here, the same set of data is repeated a finite number of times.

This is the expected behaviour in that specific case. This seems weird at first, but the structure of the input is weird here. I guess that the input comes from a real-world stream. But reality can be weird sometimes.

First, let's summarize how `tstables` and the `tables` plugin work. With long sections (_section_syntax_indicator_=1) and without option `--all-sections`, the sections are grouped in complete tables and successive identical tables are reported only once. A long section is usually uniquely identified by four different values: table id, table id extension, version number and section number. A table is identified by the first three values. On the same PID, as long as these three values are identical, no new table is reported. When _any_ of these values is different, then this is the characteristics of a new table and it is reported.

In an SDT, the table id extension is the transport stream id.

From your `sdt.bin` file, we see multiple occurrences of tables with the same set of table id, table id extension, version number. But now let's have a look at the _sequence_ of these tables:

~~~
$ tspacketize sdt.bin --pid 17 | tstables --xml - | grep 'transport_stream_id="0x0003"'
  <SDT version="11" current="true" transport_stream_id="0x0003" original_network_id="0x1111" actual="false">
  <SDT version="31" current="true" transport_stream_id="0x0003" original_network_id="0x0001" actual="false">
  <SDT version="11" current="true" transport_stream_id="0x0003" original_network_id="0x1111" actual="false">
  <SDT version="31" current="true" transport_stream_id="0x0003" original_network_id="0x0001" actual="false">
  <SDT version="11" current="true" transport_stream_id="0x0003" original_network_id="0x1111" actual="false">
  ...
~~~

You can see that no _successive_ tables have the same set of table id, table id extension, version number. For the SDT with table id extension `0x0003`, we alternate between versions 11 and 31 of the table. So, each table is considered as a change from the previous one and it is reported.

Looking closer, we can see that the original network id is different. So, we are in the weird case of a network having several TS with the _same TS id_. It seems that the network has merged several TS from different original networks, some of them having the same TS id. This is a DVB corner case. On a DVB standpoint, we understand the intention. On a practical standpoint, this is a situation where many things can go wrong:

- When an STB or TV set scans a network and finds several TS with the same TS id, most devices may be lost. Specifically, in the case of a DVB-T network, the device may conclude that the two TS are identical but come from distinct transmitters and it will keep only the one with the best signal quality. Well-behaved DVB devices will detect the different original network ids but I have learnt, by experience, to never rely on the quality of STB middleware...

- This situation defeats the usual MPEG rules for the notification of updated tables. The same set of two or more interleaved "SDT Other" from distinct original networks are repeatedly reported. This is what happens with TSDuck.

To fix that, `tstables` (and more generally the section demux in TSDuck) should understand the semantics of the various sections and, in some cases such as the DVB SDT, adopt specific detection rules. This is possible but never complete in a generic toolset such as TSDuck, working also in non-DVB MPEG environments. So, it will probably remains like that.

milos-sedlacek, 2019-05-17:
Hello, no problem with that :) As I said I can live with that.
This is satellite case and we can not influence the TS ID, ON ID of the other streams from foreign networks we reference in our SDTo. That means I don't think this at the edge of DVB. 
The implementation is your decision and I accept it.

closed 2019-05-17 by lelegard