tsduck/tsduck #300: How to restamp PTS based on another pid?

lars18th, 2019-05-21:
Hi @lelegard ,

I have this problem: I'm trying to process a program that it's sharing the teletext service between different programs insdide the same MPTS. The time difference between the PTS marks in teletext and in my target program is around 24 hours. I suspect that the clock of the target program has raised the top, but not the clock of the teletext. In any  case, the problem is that I want to recalculate the clock of the teletext (other programs aren't relevant to me), based on the PTS marks from the video of my target program. However, I don't know how to do it with a TSDuck plugin.

Any idea?
Regards.

lelegard, 2019-05-21:
I don't find a solution to that kind of problem with exiting plugins. This is a very specific need but it could be relatively easily implemented in a new plugin.

You find the service, its PMT, its video and teletext PID's. Then read video PTS and adjust the PTS in the teletext PID accordingly, taking into account the bitrate and packet distance. This won't be accurate since PTS are not PCR, they are out of sequence. But a few milliseconds do not make any noticeable difference in subtitles.

Considering you have implemented the PES mode in the `encap` /  `decap` plugins, I am sure you can do this one.

lars18th, 2019-05-21:
> I don't find a solution to that kind of problem with exiting plugins. This is a very specific need but it could be relatively easily implemented in a new plugin.

Ok. :cry: 

> You find the service, its PMT, its video and teletext PID's. Then read video PTS and adjust the PTS in the teletext PID accordingly, taking into account the bitrate and packet distance. This won't [...]

But to overcome the time drift, I think it's better a simple "offset" over the original value. Do you think so? Then for a "new" plugin `ptsrestamp` the required parameters can be:
- `--reference-pts [pid]`: indicates the pid with the PTS to compensate.
- `--pid [pid]-[pid]`: the pids to restamp.
- `--offset [value]`: as an option a value (positive/negative) to apply manually to the calculated difference.

And the work to do will be: for each target pid calculate the offset with the `reference-pts` and apply this calculated offset to all PTS/DTS values for every target pid (with the corresponding own offset).

What you think?

lars18th, 2019-07-03:
Hi @lelegard ,

Well, when the new `retime` plugin will be merged (#347), I'll try to implement this enhancement. Please, comment your opinion:

- `-P retime --master-offset-label L1 --master-pid BBB --master-offset-label L2 -p XXX`

The idea is this: the `retime` plugin adds three new parameters. The `--maste-pid` is the pid with the reference to synchronize. And the other two are the labels that will be set in the correct moment to read the PTS values of each of the two sub-streams to calculate the synch. As you can imagine the idea is to trigger the labels when a known event appears in both sub-streams (aka the synch-point). So, when the L1 and L2 events appears, then we can calculate the PTS difference, and apply the offset.

What you think?

lars18th, 2021-03-10:
Hi,

I'm also closing this other topic as it is related to this other one: https://github.com/tsduck/tsduck/issues/668#issuecomment-795066446

closed 2021-03-10 by lars18th