tsduck/tsduck #305: tsPCRAnalyzer.cpp is not compiling with gcc 9.1 under Ubuntu

haronix-bg, 2019-05-27:
Dear @lelegard 
I upgraded my GCC to version 9.1.0, but when I compiled the sources with make it has stopped with the folllowing messages:

make CC=gcc-9.1 GCC=gcc-9.1 CXX=g++-9.1 AR=gcc-ar-9.1 NODTAPI=1
  [CXX] tsPCRAnalyzer.cpp
tsPCRAnalyzer.cpp: In constructor 'ts::PCRAnalyzer::PCRAnalyzer(size_t, size_t)':
tsPCRAnalyzer.cpp:46:1: error: 'ts::PCRAnalyzer::_packet_pcr_index_map' should be initialized in the member initialization list [-Werror=effc++]
   46 | ts::PCRAnalyzer::PCRAnalyzer(size_t min_pid, size_t min_pcr) :
      | ^~
cc1plus: all warnings being treated as errors
../../Makefile.common:898: recipe for target 'release-x86_64/tsPCRAnalyzer.o' failed
make[2]: *** [release-x86_64/tsPCRAnalyzer.o] Error 1
 
I have some suggestions for this error: in line 46 all values are declared without _packet_pcr_index_map. With which one value I have to add this starting variable?

Best regards

lelegard, 2019-05-27:
Hi @haronix-bg 

Thank you for reporting this. The member field `_packet_pcr_index_map` was added in a very recent code contribution. The addition was not tested yet with all compilers and versions. This is fixed now.

closed 2019-05-27 by lelegard

haronix-bg, 2019-05-27:
Dear @lelegard 
Thanks for the fast reply and quickly fix. I'm not saw the bug fix from yesterday. I'm sorry that loss your time with my stupid ticket.