tsduck/tsduck #308: Question: is it possible to retrieve PID list order in PMT?

apallier, 2019-05-28:
PMT describes service and component info.
The list of components (given by TSDuck output) seems to be sorted by numerical order of PID.
Is there a way to get the order defined by PMT in the stream?

lelegard, 2019-05-28:
Hi @apallier,

If you mean using TSDuck commands such as `tstables` or `tstabdump`, the answer is yes. They dump the content of tables and sections as they are. So, here, you can see the order of PIDs and descriptors. If you see PIDs in numerical order, this is because the original equipment generated the table this way.

If you mean in a programatic way, the answer is no at high level and yes at low-level. C++ classes such as `ts::PMT` describe objects which can be serialized and deserialized in a number of ways. The abstract representation of the object is independent of the many ways it can be serialized. If you need to analyze this programatically, then you should collect binary tables or sections, (class `ts::Section`) and analyze the section payload yourself.

apallier, 2019-05-28:
Hi @lelegard Thank you for you quick feedback.

**First case:**

I use this command line:
`tsp -I ip 233.60.80.212:5000 -P tables --pid 1360 -P until -s 2 -O drop`

Result is good, order is preserved in the output (actually it's exactly what I expect [1369, 1364, 1368, 1367, 1366]):
```
...
Elementary stream: type 0x1B (AVC video), PID: 1369 (0x0559)
Elementary stream: type 0x06 (MPEG-2 PES private data), PID: 1364 (0x0554)
Elementary stream: type 0x11 (MPEG-4 AAC Audio), PID: 1368 (0x0558)
Elementary stream: type 0x06 (MPEG-2 PES private data), PID: 1367 (0x0557)
Elementary stream: type 0x11 (MPEG-4 AAC Audio), PID: 1366 (0x0556)
...
```

**Second case:**

But with this command line:
`tsp -I ip 233.60.80.212:5000 -P tables --pid 1360 --xml-ouput output_pmt.xml -P until -s 2 -O drop`

Result is bad. Component list seems to be ordered by elementary_PID:

```xml
<tsduck>
  <PMT version="1" current="true" service_id="0x0550" PCR_PID="0x0559">
    <!-- PID 0x0550 (1360) -->
    <registration_descriptor format_identifier="0x43554549"/>
    <component elementary_PID="0x0554" stream_type="0x06">
      <teletext_descriptor>
        <teletext language_code="swe" teletext_type="0x01" page_number="100"/>
        <teletext language_code="swe" teletext_type="0x02" page_number="799"/>
        <teletext language_code="swe" teletext_type="0x05" page_number="199"/>
        <teletext language_code="dan" teletext_type="0x02" page_number="591"/>
      </teletext_descriptor>
    </component>
    <component elementary_PID="0x0556" stream_type="0x11">
      <ISO_639_language_descriptor>
        <language code="swe" audio_type="0x03"/>
      </ISO_639_language_descriptor>
      <supplementary_audio_descriptor mix_type="0" editorial_classification="0x03"/>
      <AAC_descriptor profile_and_level="0x58" SAOC_DE="false">
        <additional_info>
          42
        </additional_info>
      </AAC_descriptor>
      <MPEG4_audio_descriptor MPEG4_audio_profile_and_level="0x58"/>
    </component>
    <component elementary_PID="0x0557" stream_type="0x06">
      <ISO_639_language_descriptor>
        <language code="swe" audio_type="0x00"/>
      </ISO_639_language_descriptor>
      <AC3_descriptor component_type="0x44"/>
      <AC3_descriptor component_type="0x42"/>
    </component>
    <component elementary_PID="0x0558" stream_type="0x11">
      <ISO_639_language_descriptor>
        <language code="swe" audio_type="0x00"/>
      </ISO_639_language_descriptor>
      <AAC_descriptor profile_and_level="0x58" SAOC_DE="false">
        <additional_info>
          03
        </additional_info>
      </AAC_descriptor>
      <MPEG4_audio_descriptor MPEG4_audio_profile_and_level="0x58"/>
    </component>
    <component elementary_PID="0x0559" stream_type="0x1B"/>
  </PMT>
</tsduck>
```

I may missed an option to have the right order.

lelegard, 2019-05-28:
OK, you have two cases here.

The first case is a dump of a section and you see the actual order of PID's in this section.

The second case is an XML object which is similar to the C++ object I mentioned. It is an abstract representation of a table. The semantics is the same as the section. But the order is not necessarily the same. In fact, the C++ classes are the central pivot format for all tables and descriptors. Binary tables are serialized and deserialized from and to C++ objects. The same applied to XML tables.

The real question is : does it matter ?

If you want to analyze the actual structure of an existing transmission, use the textual dump format.

If you want to modify a table, use the XML format. Then, I agree that the re-serialized PMT section will have its components in a possibly different order as the original one. But does it matter? All conforming receivers shall equally treat a PMT, regardless of the order of PID's.

apallier, 2019-05-28:
Yes it matters for me because I build a tool based on the XML output and I especially check components order in PMT :)
I assumed that XML output was as reliable as Text output.
Thank you for your answer. I close the issue.

closed 2019-05-28 by apallier