tsduck/tsduck #309: TSP: Stream a Tokenized HLS input

Sefy76, 2019-05-29:
I've got an HLS input which has a Token in order to play it
This works great with VLC and FFMPEG
But when I try to use TSP -I hls
Then it gives: Error: hls: invalid HLS playlist, does not start with #EXTM3U

Which I believe it's simply not taking the Token HLS URL
Is there anyway to make this work like with ffmpeg and vlc ?

lelegard, 2019-05-29:
The `hls` input plugin is a rudimentary gateway between the HLS and TS worlds. It does not perform any HLS encryption or authentication. It works only on clear and open streams. It also only works on TS content, not fragmented MP4.

Sefy76, 2019-05-29:
From doing a "dump" using ffmpeg, it seems like the source is a TS
Are there any details I can give you (other then the stream, as it's private, can't put it on public) to assist?

lelegard, 2019-05-29:
> Are there any details I can give you (other then the stream, as it's private, can't put it on public) to assist?

Please provide the complete `tsp` command with debug output. The command should be something like:
~~~
tsp --debug -I hls ... -O drop
~~~

Sefy76, 2019-05-29:
I did the command like you wrote, here is the result (minus the source name, i can't distribute it)

> tsp --debug -I hls https://<url with token> -O drop
tsp: Debug: debug level set to 1
* Debug: debug level set to 1
tsp: Debug: trying to load C:\Actus\Bin\tsplugin_hls.dll
tsp: Debug: symbol tspNewProcessor not found in C:\Actus\Bin\tsplugin_hls.dll
tsp: Debug: trying to load C:\Actus\Bin\tsplugin_drop.dll
tsp: Debug: symbol tspNewInput not found in C:\Actus\Bin\tsplugin_drop.dll
tsp: Debug: symbol tspNewProcessor not found in C:\Actus\Bin\tsplugin_drop.dll
tsp: Debug: realtime: yes, using --max-input-packets 1,000 --max-flushed-packets 1,000
* Debug: hls: debug level set to 1
* Debug: drop: debug level set to 1
* Debug: tsp: buffer size: 89,240 TS packets, 16,777,120 bytes
* Debug: hls: downloading https://<url with token>
* Debug: hls: HTTP header: HTTP/1.1 200 OK
* Debug: hls: HTTP header: Server: Apache
* Debug: hls: HTTP header: ETag: "6937b01271a16ff0041e298c8e7b6960:1559113297"
* Debug: hls: HTTP header: Last-Modified: Wed, 29 May 2019 07:01:37 GMT
* Debug: hls: HTTP header: Accept-Ranges: bytes
* Debug: hls: HTTP header: Content-Length: 2077
* Debug: hls: announced content size: 2077 bytes
* Debug: hls: HTTP header: Expires: Wed, 29 May 2019 10:04:16 GMT
* Debug: hls: HTTP header: Cache-Control: max-age=0, no-cache, no-store
* Debug: hls: HTTP header: Pragma: no-cache
* Debug: hls: HTTP header: Date: Wed, 29 May 2019 10:04:16 GMT
* Debug: hls: HTTP header: Connection: keep-alive
* Debug: hls: HTTP header: Set-Cookie: hdnea=exp=1559210655~acl=%2f*~hmac=748a4e32e47ea4eebac08163bb72f07baa2de93bf732675b895ad7f165ce5e72; path=/; domain=fncgohls-i.akamaihd.net;
* Debug: hls: HTTP header: Set-Cookie: _alid_=93WaRySEs7fXndmlShU7nw==; path=/hls/live/263399/FNCGOHLSv2/; domain=fncgohls-i.akamaihd.net
* Debug: hls: HTTP header: Access-Control-Allow-Headers: origin,range,hdntl,hdnts
* Debug: hls: HTTP header: Access-Control-Expose-Headers: Server,range,hdntl,hdnts
* Debug: hls: HTTP header: Access-Control-Allow-Methods: GET, HEAD, OPTIONS
* Debug: hls: HTTP header: Access-Control-Allow-Origin: *
* Debug: hls: HTTP header: Content-Type: application/x-mpegURL
* Debug: hls: MIME type: application/x-mpegurl
* hls: downloaded *~hmac=d36f0c1417d15f68746c95c852fcaf958060fdd6e905b5b3cfe91a850f9a93ca, master playlist, static, 8 media playlists
* hls: selected playlist: https://<....>Fox_News_v2_VIDEO_0_2564000.m3u8, 1280x720, 3,321,588 b/s, @60 fps
* Debug: hls: downloading https://<....>Fox_News_v2_VIDEO_0_2564000.m3u8
* Debug: hls: HTTP header: HTTP/1.1 403 Forbidden
* Debug: hls: HTTP header: Server: AkamaiGHost
* Debug: hls: HTTP header: Mime-Version: 1.0
* Debug: hls: HTTP header: Content-Length: 175
* Debug: hls: announced content size: 175 bytes
* Debug: hls: HTTP header: Expires: Wed, 29 May 2019 10:04:16 GMT
* Debug: hls: HTTP header: Cache-Control: max-age=0, no-cache, no-store
* Debug: hls: HTTP header: Pragma: no-cache
* Debug: hls: HTTP header: Date: Wed, 29 May 2019 10:04:16 GMT
* Debug: hls: HTTP header: Connection: keep-alive
* Debug: hls: HTTP header: X-Object-Too-Old: false
* Debug: hls: HTTP header: Access-Control-Allow-Headers: origin,range,hdntl,hdnts
* Debug: hls: HTTP header: Access-Control-Expose-Headers: Server,range,hdntl,hdnts
* Debug: hls: HTTP header: Access-Control-Allow-Methods: GET, HEAD, OPTIONS
* Debug: hls: HTTP header: Access-Control-Allow-Origin: *
* Debug: hls: HTTP header: Content-Type: application/x-mpegURL
* Debug: hls: MIME type: application/x-mpegurl
* Error: hls: invalid HLS playlist, does not start with #EXTM3U
* hls: selected playlist: https://<....>Fox_News_v2_VIDEO_0_2564000.m3u8, 1280x720, 3,321,588 b/s, @60 fps
* Debug: hls: downloading https://<....>Fox_News_v2_VIDEO_0_2564000.m3u8
* Debug: hls: HTTP header: HTTP/1.1 403 Forbidden
* Debug: hls: HTTP header: Server: AkamaiGHost
* Debug: hls: HTTP header: Mime-Version: 1.0
* Debug: hls: HTTP header: Content-Length: 175
* Debug: hls: announced content size: 175 bytes
* Debug: hls: HTTP header: Expires: Wed, 29 May 2019 10:04:16 GMT
* Debug: hls: HTTP header: Cache-Control: max-age=0, no-cache, no-store
* Debug: hls: HTTP header: Pragma: no-cache
* Debug: hls: HTTP header: Date: Wed, 29 May 2019 10:04:16 GMT
* Debug: hls: HTTP header: Connection: keep-alive
* Debug: hls: HTTP header: X-Object-Too-Old: false
* Debug: hls: HTTP header: Access-Control-Allow-Headers: origin,range,hdntl,hdnts
[... 103 lines cut ...]
* Debug: hls: HTTP header: Content-Type: application/x-mpegURL
* Debug: hls: MIME type: application/x-mpegurl
* Error: hls: invalid HLS playlist, does not start with #EXTM3U
* hls: selected playlist: https://<....>Fox_News_v2_VIDEO_3_364000.m3u8, 256x144, 508,358 b/s, @30 fps
* Debug: hls: downloading https://<....>Fox_News_v2_VIDEO_3_364000.m3u8
* Debug: hls: HTTP header: HTTP/1.1 403 Forbidden
* Debug: hls: HTTP header: Server: AkamaiGHost
* Debug: hls: HTTP header: Mime-Version: 1.0
* Debug: hls: HTTP header: Content-Length: 175
* Debug: hls: announced content size: 175 bytes
* Debug: hls: HTTP header: Expires: Wed, 29 May 2019 10:04:16 GMT
* Debug: hls: HTTP header: Cache-Control: max-age=0, no-cache, no-store
* Debug: hls: HTTP header: Pragma: no-cache
* Debug: hls: HTTP header: Date: Wed, 29 May 2019 10:04:16 GMT
* Debug: hls: HTTP header: Connection: keep-alive
* Debug: hls: HTTP header: X-Object-Too-Old: false
* Debug: hls: HTTP header: Access-Control-Allow-Headers: origin,range,hdntl,hdnts
* Debug: hls: HTTP header: Access-Control-Expose-Headers: Server,range,hdntl,hdnts
* Debug: hls: HTTP header: Access-Control-Allow-Methods: GET, HEAD, OPTIONS
* Debug: hls: HTTP header: Access-Control-Allow-Origin: *
* Debug: hls: HTTP header: Content-Type: application/x-mpegURL
* Debug: hls: MIME type: application/x-mpegurl
* Error: hls: invalid HLS playlist, does not start with #EXTM3U
* Error: hls: no more media playlist to try, giving up
* Debug: Report logging thread terminated

lelegard, 2019-05-29:
~~~
* Debug: hls: HTTP header: Content-Type: application/x-mpegURL
* Debug: hls: MIME type: application/x-mpegurl
* hls: downloaded *~hmac=d36f0c1417d15f68746c95c852fcaf958060fdd6e905b5b3cfe91a850f9a93ca, master playlist, static, 8 media playlists
* hls: selected playlist: https://<....>Fox_News_v2_VIDEO_0_2564000.m3u8, 1280x720, 3,321,588 b/s, @60 fps
* Debug: hls: downloading https://<....>Fox_News_v2_VIDEO_0_2564000.m3u8
* Debug: hls: HTTP header: HTTP/1.1 403 Forbidden
* Debug: hls: HTTP header: Server: AkamaiGHost
....
* Error: hls: no more media playlist to try, giving up
* Debug: Report logging thread terminated
~~~

 The akamai server returned a master playlist with 8 media playlists, typically with different bitrates and resolutions. The `hls` plugin tries them all, one by one, and they all fail with status `403 Forbidden`.

The master playlist is returned with 2 cookies. These cookies are currently not sent with the subsequent requests for media playlists. This may be the cause for the access denial. But, even with the cookies, I do not know if this would allow the TS download in case of token encryption.

I can make the modification but I need to test on the actual stream.

Sefy76, 2019-05-29:
I can give you the actual URL, as long as it's not publicized here. (can I private message?)
Note that both VLC and FFMPEG are opening the URL without any problem.
I think the 403 Forbidden is occurring, as TSP is not inserting the Token for the HLS

lelegard, 2019-05-29:
Thanks for the link. It seems that there is only a cookie-based authentication but no encryption.

I added the transmission of the cookies from the master playlist to media playlists and to media segments. It now works. Propagating the cookies was sufficient to perform authentication.

Sefy76, 2019-05-29:
Oh! That was quick! Thank you!
So is there a new build or his dll that I can get ?

lelegard, 2019-05-29:
> So is there a new build or his dll that I can get ?

No. Only official releases have published binaries. The next one will be published somewhere in June.

To get the latest updates, you need to build TSDuck yourself [as explained here](https://tsduck.io/doxy/building.html). This is easy. Many users rebuild TSDuck routinely.

Sefy76, 2019-05-30:
I'm not a developer or anything like that
Never build a project and I don't have the tools.

Is there any chance you can just send me the fixed file you have? June is
just around the corner anyway :)

lelegard, 2019-05-30:
OK, I pushed some fresh binaries here: https://tsduck.io/download/prerelease/

From the debug traces you posted, I assume you are working on Windows.

Sefy76, 2019-05-30:
Thank you ever so much for the help! much appreciated! :)

closed 2019-05-30 by Sefy76