tsduck/tsduck #310: Plugin filter: cutoff (UPD receiver)

lars18th, 2019-05-30:
Hi @lelegard ,

Still waiting to test the new plugin `trigger` (I'm very busy, sorry!)... #294 .
But regarding the functionality of sending a signal using UDP datagrams, why not implement the opposite? That's a plugin like `cutoff` that can receive UDP datagrams (as other plugins already have implemented) and generate TSDuck signals (setting labels)?

With these two filters (`trigger` and `cutoff`) will be then possible to build a full external controller for a TSDuck process.

What you think?

lelegard, 2019-05-30:
Plugin `cutoff` added. This one was easy, all the building blocks were already there.

I tested it on macOS only. Please test in your environment.

lars18th, 2019-05-31:
> Plugin `cutoff` added. This one was easy, all the building blocks were already there.

Hey, man! You're a superhero and this project is amazing!!! :clinking_glasses: 
Thanks a lot! :stuck_out_tongue_winking_eye:

rezna11, 2019-05-31:
Sometimes with all these new plugins it would be nice to share a real-world example what are they used for, unless it's a secret :)

lelegard, 2019-05-31:
> Sometimes with all these new plugins it would be nice to share a real-world example what are they used for, unless it's a secret :)

Good point indeed. The case of `trigger` / `cutoff` was made possible with the addition of the packet labelling.

The basic blocks are:
- Traditional plugins working on packets.
- New plugins working on labels (`trigger` / `cutoff`)
- Interaction between the two (`-P filter --set-label` and option `--only-label` on all plugins).

For instance, you can mute and restore a service in a TS in transit using a remote control, something like (untested):

~~~~
tsp -I dvb ... \
    -P cutoff ... \
    -P svremove name1 --stuffing --only-label 1 \
    -P svremove name2 --stuffing --only-label 2 \
    -O dektec ...
~~~~

By sending `"start-label 1"` and `"stop-label 1"`, you mute (and replace by stuffing) and restore the service `name1`. Same for 2, etc.

lars18th, 2019-05-31:
Hi @rezna11 ,

I commented it!

> build a full external controller for a TSDuck process.

Perhaps at some time I'll work on adding support for ZeroMQ (instead of the current plain UDP support).

lars18th, 2019-05-31:
Hi @lelegard ,

> For instance, you can mute and restore a service in a TS in transit using a remote control...

Yes, "mute" is one option. But for "pause" we have some option?

Think on this environment:
- We have several inputs.
- Then we use `tsswitch` for selecting between the inputs.
- The final output is sended over a multicast (streaming).

Now we have all tools to control this environment. However, if we like to _PAUSE_ the inputs what's the method? Reasons to want to do that:
- The inputs are HLS and we don't like to get the stream if the input isn't selected.
- The inputs are PIPEs from one stream processor (encoder) and we don't like to encode if the signal isn't used.
- The inputs are MULTCASTS and we don't require to receive the UDP packets if the input isn't readed.

Any idea?

rezna11, 2019-05-31:
> Hi @rezna11 ,
> 
> I commented it!
> 
> > build a full external controller for a TSDuck process.
> 
> Perhaps at some time I'll work on adding support for ZeroMQ (instead of the current plain UDP support).

I know but I was just curious, for example in what situation would you use TSDuck in such remote-controlled way. There's always a brief technical description of what a plugin does but sometimes it's hard to imagine a real-life usage to inspire others to use it. Basically stuff that's covered in the great chapter 5 of the TSDuck manual... but sorry for going off topic, you can get back to designing great new plugins now :)

lelegard, 2019-05-31:
> Perhaps at some time I'll work on adding support for ZeroMQ (instead of the current plain UDP support).

ZeroMQ is one of the many distributed system middleware that existed such as plain old RPC, CORBA, Dbus, etc. This kind of software comes and goes, just like fashion. I do not want to create a dependency on a such libraries.

> However, if we like to PAUSE the inputs what's the method?

I suppose you mean a _time shift_ feature as implemented on most set-top boxes. This requires the same infrastructure as a PVR. I do not think that TSDuck is the best tools for this.

lars18th, 2019-05-31:
Hi @rezna11 ,

> it's hard to imagine a real-life usage to inspire others to use it... but sorry for going off topic.

Not in my opinion. I like to discuss about ideas and real-life usage. :wink: 
Just my two cents:
- Sometimes it's quite hard to use some _library_ with an API. Why? Because in your environment perhaps you are using some scripting language and it's difficult to create/use a binding. In this case, then you can use _programs_ (processes) and _control_ them. And now with the TSDuck you can launch different `tsp` processes and send/receive messages from them. So it's quite simple to build a complex project without touching the C/C++ API of TSDuck.

lars18th, 2019-05-31:
Hi @lelegard ,

> > However, if we like to PAUSE the inputs what's the method?
> 
> I suppose you mean a _time shift_ feature as implemented on most set-top boxes. This requires the same infrastructure as a PVR. I do not think that TSDuck is the best tools for this.

No, no, sorry!
For "PAUSE" I don't request about _time shift_ at all.
I comment about **pausing** the execution of the `tsp` **loop**! Example:
```
$ tsp -I hls http://server:/stream-1.hls -O ip 224.0.0.1:1234

$ tsp -I hls http://server:/stream-2.hls -O ip 224.0.0.2:1234

$ tsswitch --remote 4444 -I ip 224.0.0.1:1234 -I ip 224.0.0.1:1234 | tsp -O ip 226.1.1.1:1234
```

The objective here it's to **pause** the HLS of the unused input. That's stopping to receive the chunks until the input is selected. More or less a "pause in the loop".
Any idea?

And regarding...

> ZeroMQ is one of the many distributed system middleware that existed such as plain old RPC, CORBA, Dbus, etc. This kind of software comes and goes, just like fashion. I do not want to create a [...]

Yes, I agree. I'm only speaking about "at some time" _expand_ the current UDP implementation with some code to send/receive ZeroMQ messages without using external translators (UDP-2-ZMQ). For sure the idea is with something optional (like with `NOPCSC=1` but disabled by default: `make ZMQ=1`). 

Don't worry about it, it's a personal idea!

Regards.

closed 2019-07-04 by lelegard