tsduck/tsduck #313: Testing TSDuck 3.18-1283 on Windows (Visual Studio 2019)

lelegard, 2019-06-01:
Hi all,

It is time to move forward in terms of build tools on Windows. I have updated the MSVC project files so that TSDuck can now be built using Visual Studio 2019. Additionally, the installer is now built using NSIS version 3.04 (was 2.46 for years). Note that the MSVC project files were not simply upgraded by Visual Studio (it does a dirty job), they were carefully manually modified to be compatible with Visual Studio 2015, 2017 and 2019. Or at least I hope they are...

The installers and binaries work on the latest version of Windows 10. I would like to test something for which I need your collaboration:

1. Does the binary still work correctly on older versions of Windows, down to Windows 7?
2. Does the build still work with Visual Studio 2017 or even 2015?

Windows users, could you please test that the binaries work on older (but not too old) Windows versions? The pre-release version 3.18-1283 is available [here](https://tsduck.io/download/prerelease/).

Windows developers with Visual Studio 2015 or 2017, could you please pull the repo and test the build? 

Important: For developers, the directory `build\msvc2017` has been renamed `build\msvc` since it supports multiple versions of Visual Studio and keeping 2017 in the name is misleading.

Please provide your feedback here, including your versions of Windows (all) and Visual Studio (developers). Thanks in advance for the collaborative testing work.

Sefy76, 2019-06-03:
I've been working with it on both Windows 7, Windows 10 and Server 2008 and 2012/2016 and i've not had any issues so far.

closed 2019-06-25 by lelegard

oviano, 2019-08-13:
Ok, so the latest release 3.18-1361 does not build with VS2017 on Windows 10. All the projects reference platform toolset 142, which is not part of VS2017.

So I changed all the projects back top v141, but then I get these build errors:

`1>------ Build started: Project: tsducklib, Configuration: Release x64 ------
1>tsDektecDevice.cpp
1>tsDektecVPD.cpp
1>tsAIT.cpp
1>tsAbstractDescrambler.cpp
1>tsAbstractDuplicateRemapPlugin.cpp
1>tsAbstractHTTPInputPlugin.cpp
1>tsAbstractTablePlugin.cpp
1>tsAbstractTransportListTable.cpp
1>tsArgs.cpp
1>tsArgsWithPlugins.cpp
1>tsAudioLanguageOptions.cpp
1>tsBAT.cpp
1>tsCASMapper.cpp
1>tsCASSelectionArgs.cpp
1>tsCVCT.cpp
1>tsDektecControl.cpp
1>c:\program files (x86)\microsoft visual studio\2017\community\vc\tools\msvc\14.16.27023\include\xtree(416): error C2220: warning treated as error - no 'object' file generated (compiling source file ..\..\src\libtsduck\tsAbstractTransportListTable.cpp)
1>c:\program files (x86)\microsoft visual studio\2017\community\vc\tools\msvc\14.16.27023\include\xtree(59): note: see reference to class template instantiation 'std::_Tree_node<_Ty,std::_Default_allocator_traits<_Alloc>::void_pointer>' being compiled
1>        with
1>        [
1>            _Ty=std::pair<const ts::TransportStreamId,ts::AbstractTransportListTable::Transport>,
1>            _Alloc=std::allocator<std::pair<const ts::TransportStreamId,ts::AbstractTransportListTable::Transport>>
1>        ] (compiling source file ..\..\src\libtsduck\tsAbstractTransportListTable.cpp)
1>c:\program files (x86)\microsoft visual studio\2017\community\vc\tools\msvc\14.16.27023\include\xtree(58): note: while compiling class template member function 'std::_Tree_unchecked_const_iterator<std::_Tree_val<std::_Tree_simple_types<_Ty>>,std::_Iterator_base0> &std::_Tree_unchecked_const_iterator<std::_Tree_val<std::_Tree_simple_types<_Ty>>,std::_Iterator_base0>::operator ++(void)'
1>        with
1>        [
1>            _Ty=std::pair<const ts::TransportStreamId,ts::AbstractTransportListTable::Transport>
1>        ] (compiling source file ..\..\src\libtsduck\tsAbstractTransportListTable.cpp)
1>c:\program files (x86)\microsoft visual studio\2017\community\vc\tools\msvc\14.16.27023\include\xtree(240): note: see reference to function template instantiation 'std::_Tree_unchecked_const_iterator<std::_Tree_val<std::_Tree_simple_types<_Ty>>,std::_Iterator_base0> &std::_Tree_unchecked_const_iterator<std::_Tree_val<std::_Tree_simple_types<_Ty>>,std::_Iterator_base0>::operator ++(void)' being compiled
1>        with
1>        [
1>            _Ty=std::pair<const ts::TransportStreamId,ts::AbstractTransportListTable::Transport>
1>        ] (compiling source file ..\..\src\libtsduck\tsAbstractTransportListTable.cpp)
1>c:\program files (x86)\microsoft visual studio\2017\community\vc\tools\msvc\14.16.27023\include\xtree(196): note: see reference to class template instantiation 'std::_Tree_unchecked_const_iterator<std::_Tree_val<std::_Tree_simple_types<_Ty>>,std::_Iterator_base0>' being compiled
1>        with
1>        [
1>            _Ty=std::pair<const ts::TransportStreamId,ts::AbstractTransportListTable::Transport>
1>        ] (compiling source file ..\..\src\libtsduck\tsAbstractTransportListTable.cpp)
1>c:\program files (x86)\microsoft visual studio\2017\community\vc\tools\msvc\14.16.27023\include\xtree(416): error C2220: warning treated as error - no 'object' file generated (compiling source file ..\..\src\libtsduck\tsCASSelectionArgs.cpp)
1>c:\program files (x86)\microsoft visual studio\2017\community\vc\tools\msvc\14.16.27023\include\xtree(59): note: see reference to class template instantiation 'std::_Tree_node<_Ty,std::_Default_allocator_traits<_Alloc>::void_pointer>' being compiled
1>        with
1>        [
1>            _Ty=std::pair<const ts::PID,ts::PMT::Stream>,
1>            _Alloc=std::allocator<std::pair<const ts::PID,ts::PMT::Stream>>
1>        ] (compiling source file ..\..\src\libtsduck\tsCASSelectionArgs.cpp)
1>c:\program files (x86)\microsoft visual studio\2017\community\vc\tools\msvc\14.16.27023\include\xtree(309): note: see reference to class template instantiation 'std::_Tree_const_iterator<std::_Tree_val<std::_Tree_simple_types<_Ty>>>' being compiled
1>        with
1>        [
1>            _Ty=std::pair<const ts::TransportStreamId,ts::AbstractTransportListTable::Transport>
1>        ] (compiling source file ..\..\src\libtsduck\tsAbstractTransportListTable.cpp)
1>c:\program files (x86)\microsoft visual studio\2017\community\vc\tools\msvc\14.16.27023\include\xtree(58): note: while compiling class template member function 'std::_Tree_unchecked_const_iterator<std::_Tree_val<std::_Tree_simple_types<_Ty>>,std::_Iterator_base0> &std::_Tree_unchecked_const_iterator<std::_Tree_val<std::_Tree_simple_types<_Ty>>,std::_Iterator_base0>::operator ++(void)'
1>        with
1>        [
1>            _Ty=std::pair<const ts::PID,ts::PMT::Stream>
1>        ] (compiling source file ..\..\src\libtsduck\tsCASSelectionArgs.cpp)
1>c:\users\oliver\perforce\non-si\thirdparty\sdks\tsduck\custom\src\libtsduck\tsabstracttransportlisttable.cpp(93): note: see reference to class template instantiation 'std::_Tree_iterator<std::_Tree_val<std::_Tree_simple_types<_Ty>>>' being compiled
1>        with
1>        [
1>            _Ty=std::pair<const ts::TransportStreamId,ts::AbstractTransportListTable::Transport>
1>        ]
[... cut ...]
1>c:\program files (x86)\microsoft visual studio\2017\community\vc\tools\msvc\14.16.27023\include\string(656): note: see reference to class template instantiation 'std::basic_string<char16_t,std::char_traits<char16_t>,std::allocator<char16_t>>' being compiled (compiling source file ..\..\src\libtsduck\tsPlugin.cpp)
1>c:\program files (x86)\microsoft visual studio\2017\community\vc\tools\msvc\14.16.27023\include\xtree(416): warning C4623: 'std::_Tree_node<_Ty,std::_Default_allocator_traits<_Alloc>::void_pointer>': default const