tsduck/tsduck #314: TSP: Loop an Input TS file

Sefy76, 2019-06-02:
I'm not sure if I missed it in the documents, but i've tried looking everywhere and could not find.
Is there any way of using TSP.exe to Loop an Input TS file for output Streaming?

lelegard, 2019-06-02:
With `-I file --infinite`_`filename`_ the file is played again and again. With option `--repeat`_`count`_ the file is played a given number of times.

If the output device naturally regulates the output speed (a modulator for instance), the command is simply: `tsp -I file --infinite`_`filename`_`-O dektec ...`

If the output does not regulate (`ip` or `hls` for instance), then you need in add the plugin `regulate` to ensure that the data are not sent faster than the playout speed.

See the [TSDuck user's guide](https://tsduck.io/download/docs/tsduck.pdf) sections 5.2.3 and 5.2.4.

Sefy76, 2019-06-02:
Thanks, i'll give it a try!
PS: i was looking for the word "loop" in the guide and couldn't find, but i'll check those sections!

closed 2019-06-02 by Sefy76