tsduck/tsduck #316: Scrambler a unicast stream

AndersonDan, 2019-06-09:
I'm trying to make the scrambler a unicast stream that I plan to run on Android, iOS and browsers.

Is this possible with TSDuck, or does it only support multicast streaming?

If it supports unicast, what output formats can I use, only MPEG-TS?

lelegard, 2019-06-10:
Hi,

There are two distinct layers of discussion here: transmission layer (streaming) and container layer (TS).

TSDuck is a flexible framework which manipulates MPEG Transport Streams. So, it has one container format (TS as in "TS"Duck) and many different ways or manipulating and transmitting it.

To output a TS over IP, TSDuck provides several ways: UDP and HLS. UDP can be done using multicast or unicast, it is the same, only the destination address changes (as well as some hidden details). For HLS, TSDuck generates (and purges) the files, you just need a standard Web server such as Apache to serve them. HLS is unicast by construction.

If your target is OTT clients, then you can use HLS. This is perfect for iOS and browser. I do not know anything about Android.

That being said, your question seems to describe a production situation. Please remember that TSDuck was designed from the beginning as a test, demo, integration, debug toolbox. It does not necessarily meet the robustness requirements nor supports all the fancy features for a production system. But it is free. You pay for what you get.

AndersonDan, 2019-06-11:
I do not understand exactly when you say you're not ready for production. What would be the problem, at least in all my tests I did the tsduck, it always went very well.

On the first question, I plan to get a multicast stream (TV) -> scrambler -> unicast (OTT). Did you understand?

lelegard, 2019-06-11:
> I do not understand exactly when you say you're not ready for production. What would be the problem, at least in all my tests I did the tsduck, it always went very well.

This is a matter of responsibilities and completness of features. A production tool should be designed to be _resilient_, meaning handling failover, redundancy, monitoring, reporting and much more. Most open source software, although fully functional, do not meet these requirements.

TSDuck is designed to be _robust_. And it is. But I would have designed it differently in terms of resiliency if I had had in mind to use it for production. Now, if you are happy with TSDuck, that is fine for everyone.

> On the first question, I plan to get a multicast stream (TV) -> scrambler -> unicast (OTT). Did you understand?

TSDuck can do that. But you should be more precise about "unicast output". There are several ways to do this. It can be unicast UDP (probably not what you want), HLS, DASH or even SmoothStreaming. 

> And a question that keeps stirring in my mind is that whenever I search on DRM, OTT, etc, it is never mentioned the MPEG-TS, being that this is functional.

HLS, DASH or SmoothStreaming are _protocols_. Inside the protocol, there is a _container_ format. And inside the container, there are audio/video _encoding_ formats. TS is a container format.

HLS was defined by Apple. From the beginning, TS was the unique container format inside HLS. So, yes, TS is definitely used in OTT, or some of it. Nowadays, HLS also supports fMP4 (fragmented MP4) container.

DASH does not support TS.

So, TSDuck can work for traditional HLS streaming containing TS. TSDuck cannot help you if you choose HLS with fMP4 or DASH.

closed 2019-07-31 by lelegard