tsduck/tsduck #317: Is there a sample code for using TSDuck as a library?

rezna11, 2019-06-11:
Hi, I'm trying to use TSDuck as a library in my own C++ program. There's no particular goal, I just want to see if I could use TSDuck library instead of some messy C++ code and other libraries like Poco, Boost and StreamReaderEx-BDA. But since I'm not really a developer and never had any C++ classes, I find it impossible to accomplish even a simple task like reading a TS source, modifying/analyzing packets and sending the result to output... but I feel that this should be easy with TSDuck libraries but just can't figure how to do it, so I would appreciate any sample code, if anyone has, showing how TSDuck libraries could be used.

For example, I'd like to be able to:
1) open and read a TS file, read from HTTP, read from UDP or from a tuner
2) add a PID to demux and print some of the tables as text to stdout
3) modify some PID's payload data and pass it through to output (UDP for example)

But I'm running into issues when trying to find and use the right TSDuck classes for these purposes. I've tried working with ts::TSFileInput, ts::SectionDemux, ts::WebRequest and ts::TablesLogger but running into one dead end after another, that's why I now try to ask others for their input. 

As I said, I don't really need this, since the existing TSDuck utilities with all the plugins are great for stupid people like myself who can't really code in C++ but I like to try new stuff and I thought the TSDuck library documentation is lacking some simple examples of how to use it, so I'm asking here.

I've accomplished only small stuff, like the following code, which uses TSDuck's ts::TSPacket, ts::UDPSocket and ts::TSByteBlock but the rest is just normal C++ code and although the methods like getPID() and getPUSI() are helpful, I really think it should be possible to accomplish much more with TSDuck than this, from reading the file, to printing the output, but I wasn't able to get it to work.

The following reads NIT from file and saves all NIT packets to one file, plus it sends to output to UDP.

```
void MyTSFileAnalyzer::getNIT()
{
	std::cout << "getting NIT" << std::endl;
	while (1) {
		inbuf->read(in_buffer, 188);
		if (in_buffer[0] != 0x47) { std::cout << "sync error" << std::endl; }
		else {
			in_ts_packet.copyFrom(in_buffer);
			curr_pid = in_ts_packet.getPID();
			if (curr_pid == 16 && in_ts_packet.getPUSI()) {
				sec_len = ((in_buffer[6] & 0x0f) << 8) | (in_buffer[7] & 0xff);
				printf("\nsec_len = %d", sec_len);
				full_nit.append(in_buffer, 188);
				int i = 188;
				while (i < sec_len) {
					inbuf->read(in_buffer, 188);
					in_ts_packet.copyFrom(in_buffer);
					if (in_ts_packet.getPID() == 16) {
						full_nit.append(in_buffer, 188);
						i += 188;
						printf("\nappend = %d", i);
					}
				}
				break;
			}
		}
	}
	std::ofstream outfile("new_2.ts", std::ofstream::binary);
	full_nit.write(outfile);

        ts::UDPSocket ts_sock;
	const ts::SocketAddress addr_port2(u"10.0.0.11:54321");
	const ts::SocketAddress addr_source(10, 0, 0, 11);
	ts_sock.open();
	ts_sock.bind(addr_source);
	ts_sock.setDefaultDestination(addr_port2);
	ts_sock.send(full_nit.data(), full_nit.size());
}
```

```
class MyTSFileAnalyzer
{
private:
	std::istream* inbuf;
	char in_buffer[188];
	int curr_pid;
	int sec_len;
	ts::ByteBlock full_nit;
	ts::TSPacket in_ts_packet;

public:
	MyTSFileAnalyzer(std::istream* my_inputfile);
	~MyTSFileAnalyzer();
	void getPAT();
	void getSDT();
	void getNIT();
};
```

lelegard, 2019-06-11:
There are several approaches:

- Write your application from scratch, using TSDuck classes.
- Write your application using existing plugins to perform some more generic processing.
- Write your own plugin and use `tsp` as main application.

The last case is probably the easiest way if this can match your application case. There is an example of integration of an independent third-party plugin in the TSDuck source repo in `tsduck/sample/sample-plugin`. This gives you the code skeleton. Then consider a couple of existing plugins which have a similar structure as your target. Look at their source code and adapt for your needs.

The second case is interesting, but a bit complicated. Aaron Levinson (@alevinsn) has contributed with some sample code which does this in `tsduck/sample/sample-app`. 

The first case is just plain C++.

In all cases, I am afraid it requires some familiarity with C++ object-oriented programming.

lelegard, 2019-06-11:
Considering the extraction of a table such as the NIT in your example, you should use a `ts::SectionDemux`. This is a class which does all section demuxing, table reconstruction and notifies the application when a new complete table is available. By accumulating raw TS packets as you are doing, you will have to manually recreate the complicated stuff in `ts::SectionDemux`. TSDuck classes are all designed to hide the complexity and provide a high-level and simple-to-use interface to applications or plugins.

rezna11, 2019-06-11:
Thanks, although I was trying to do what you describe in the first case, I think that the second case might be easier and the example application looks as a great starting point to give an idea how to use the plugins in C++ code.
However, I have 2 issues with the sample application code:

1) the input plugin doesn't work... the application fails to recognize "file" as the input, I think. This is the output printed when running the sample app:
```
D:\DEVEL\Projects\8 Tests\duck_sample-app\sample-app\Debug>sample-app.exe
format, i=12, s=abc
TSDuck version is: 3.17-1189
TSDuck built on: Apr 10 2019 - 22:48:23
file.dll: The specified module could not be found.
```

2) the code as provided won't compile because of ts::TSPacketMetadata, this is the error:

```
Severity	Code	Description	Project	File	Line	Suppression State
Error	C2039	'TSPacketMetadata': is not a member of 'ts'	sample-app	d:\devel\projects\8 tests\duck_sample-app\sample-app\sample-app\sample-app.cpp	178	

```
... I just created new project, added <Import Project="$(TSDUCK)\tsduck.props" />, and copy-pasted the sample code, plus I also had to add #include "stdafx.h", but I can't see why TSPacketMetadata is not found in ts::

alevinsn, 2019-06-12:
If the TSDuck library can't find a plug-in, that means it isn't available using the standard search path rules utilized by TSDuck.  Read the comment that starts on line 64 of myexec.cpp to understand this better and/or step into getInput() to see how it works (the latter implies that you'll need to build TSDuck yourself, which is a good idea anyway for this sort of development).

TSPacketMetadata corresponds to a recent change to the TSDuck library.  If it won't build, that means your TSDuck source base or SDK is out-of-date.

rezna11, 2019-06-12:
Thanks, I have built latest TSDuck to get the TSPacketMetadata working.
Being unable to find a plugin is also resolved - I installed x64 TSDuck and then try to use it in x86 application. So that's a stupid mistake. 

Now I have another problem, though. When I build the sample app for Release x64, it works. But the sample app fails to run in Debug x64, giving me an error. So I'm not sure if TSDuck can be used in debug this way but I can't imagine writing an application without being able to debug, so there should be a way...

Sorry for a screenshot but maybe you'll know what's wrong by just looking at it...

[image]

rezna11, 2019-06-12:
OK, so this is probably also easy... running `.\Build.ps1 -Debug` built the debug binaries so now I suppose I just have to set different value for working directory or ENV. variable in Visual Studio when running in Debug...

closed 2019-07-04 by lelegard

fabricioizumi, 2021-06-15:
> Considering the extraction of a table such as the NIT in your example, you should use a `ts::SectionDemux`. This is a class which does all section demuxing, table reconstruction and notifies the [...]

Where Can I find a sample code for using SectionDemux like you describe? I trying to use TSDuck as a library in my project to extract some tables from a TS file.